wled/WLED #779: Wemos D1 Mini mit Hyperion.ng

Cosmicbase, 2020-03-17:
Hallo ich habe ein Problem das meine LEDs nicht leuchten.
Wo muss ich welche Einstellungen in Hyperion.ng machen und welche in WLED?
Ich habe in Hyperion.ng Udpraw eingestellt und GBR, Port weiß ich nicht nicht welchen...
rewrite Time habe ich 1000,

Wo wird die Anzahl der LEDs, ob Rückwärts oder Vorwärts, sowie die Nummer der 1.Led angegeben?
Bei Hyperion.ng habe ich als 1. LED die 75. 
Zuvor hatte ich alles über einen Arduino Uno R3 laufen. 
Aber durch die vielen Möglichkeiten inkl. App und Alexa bin ich auf WLED umgestiegen. 
Und ja die APA102 Binary habe ich auch geflasht da ich APA102 habe. 

Bitte um Hilfe.

Aircoookie, 2020-03-18:
Hmm das sieht soweit eigentlich gut aus. Was genau passiert? Wenn zum Beispiel der WLED Solid effekt an ist, gehen die LEDs aus wenn Hyperion gestartet ist und wieder an sobald es beendet wird. Wenn das der Fall ist, funktioniert es wahrscheinlich wie es soll, nur dass Hyperion evtl. den Bildschirm nicht richtig aufnimmt. Port 19446 is auf jeden Fall richtig!

Cosmicbase, 2020-03-18:
Muss ich irgendwelche Ports im Router forwarden?
Zum Beispiel den UDP Broadcast 21234 oder 19446?

Cosmicbase, 2020-03-18:
Ich habe auch was gesehen das UDP Live steht wenn Hyperion.ng an ist und wenn aus dann WLED.

Aircoookie, 2020-03-18:
Nein, muss nichts geforwarded werden. WLED reagiert auf Pakete auf port 19446. Wenn WLED UDP Live sagt, wird das signal offenbar richtig empfangen, deswegen gehe ich eher von einem Fehler in Hyperion aus, der zur Ausgabe von einem schwarzen Bildschirm fuehrt... Gibt es da eine Livevorschau?

Cosmicbase, 2020-03-18:
Ja es gibt eine Live Vorschau. Aber WLED funktioniert ja auch ohne Hyperion.ng.
Zumindest Effekte. Aber mit meinem D1 Mini funktioniert gar nix. Ich glaube ich habe beim Pin löten eine kalte Lötstelle oder er hat einen Defekt.

Cosmicbase, 2020-03-18:
[image]
Muss ich die Weiterleitung aktivieren?
Mit IP Adresse vom WLED?

Aircoookie, 2020-03-18:
Nein, die Weiterleitung bitte nicht aktivieren.
Also wenn auch mit den internen WLED Effekten die LEDs schwarz bleiben, muss es einen Fehler mit den LEDs sein. Clock muss an D3 und Data an D4, und die Ground Anschluesse verbunden :)

Cosmicbase, 2020-03-19:
Danke

mike2nl, 2020-03-26:
> Danke

Was Danke? Funktioniert es jetzt? Dann kannst du das Issue schliessen.

D3ltorohd, 2020-06-14:
Hänge mich hier mal mit an. Hoffe das ist noch nicht zu alt. Habe das eigentlich genau so eingerichtet. Auch bei mir tut sich nichts, wenn ich in Hyperion z.b. Effekte teste. Über WLED funktionieren die LED's.

Der Einschalt Button im WLED WebIF muss auf an sein richtig ?

D3ltorohd, 2020-06-14:
So mittlerweile hab ich wohl eine Verbindung. Die LED's flackern dann immer mal kurz, die Farben sind mehrere Sekunden async und danach schaltet sich WLED immer aus, also im WEB If der Button ist aus. Wenn ich mit Hyperion NG und Screen Capture arbeite.

Paulchen-Panther, 2020-06-19:
Bitte den Weiterleitungsport auf 19447 ändern und Hyperion neustarten. Die Weiterleitung ist zwar im ausgeschalteten zustand nicht aktiv aber der Port wird belegt.
https://github.com/hyperion-project/hyperion.ng/blob/master/config/hyperion.config.json.default#L127

dermeb, 2020-07-13:
Hallo zsuammen,
@D3ltorohd klappt es bei dir? Habe genau die gleichen Symptome

D3ltorohd, 2020-07-13:
> Hallo zsuammen,
> @D3ltorohd klappt es bei dir? Habe genau die gleichen Symptome

Ja mittlerweile läuft es wunderbar. Ich musste bei den Einstellungen, einfach noch mal alles eingeben, wie z.b. den Port und erneut abspeichern. Seither klappt es wunderbar. Irgendwie scheint da noch was nicht richtig zu funktionieren. Aber nach erneutem abspeichern gings auf jeden Fall.

closed 2020-11-19 by stale[bot]