wled/WLED #780: Trying to change the LED's to white with Alexa turns the RGB values to 0

arrows94, 2020-03-17:
If I try to change the LED's to white with the voice command the values are turned to 0. It seems that it does not calculate the RGB values from the RGBW ones. It was working in 0.8.6 I think.

Aircoookie, 2020-03-18:
Just looked into the code and can confirm that this happens. Alexa white values are only optimized for RGBW strips, always setting RGB to 0,0,0 even if no White channel is present. Will make that conditional so that whites work well again with RGB.

aexeler, 2020-03-27:
Hi,
i use the Binary WLED_0.9.1_ESP32.bin with the esphome-flasher-tool and it works with the App!
Thanks for this Projet.
But i have the Alexa White Problem, how can i fix it? Modify the Binary? (and how?)
Thanks for help!
Greets
Andreas

softhack007, 2023-10-02:
Seems that this one is fixed with #3209

closed 2023-10-02 by softhack007