wled/WLED #781: [Question] Nanoleaf behaviour possible?

AnthonMS, 2020-03-18:
First of all, this project is awesome and looking so much forward to making some lamps with it.

This is not so much of an issue, but more of a question. I don't know where I can ask this except here. So hopefully that is all right, if not just close it.

I am in the process of making some DIY nanoleafs, and I really want to use this software, since it is easily integrated into Home Assistant. But I am having a hard time figuring out if it is possible to make it behave like real Nanoleafs. I know it is possible to make each panel a segment. But is it possible to make segment effects. So it will behave something like in the gif below?
[image]

If so, can you explain it to me? Because I cant for the life of me, figure out how to do this. I want to be able to select an effect that will change colors for each segment at a time, and not individual LEDs. And when I make segments, it will make seperate light entities in HA. I just want to group my LEDs up, so it changes color in groups instead of individual LEDs.

raoulteeuwen, 2020-03-18:
Nice GIF to visualize what you are looking for. Are you already using WLED with a string? Anyway, you already know WLED offers segments. You can pick any effect WLED has per segment. So you could either pick any of the available effects, or add your own. Afaik (i do have HA), every controller you use turns up as 1 WLED in HA. The only current limitation in WLED with standard code is a limit of 10 segments and only one memory location that supports storing segments and effects. You can manually override the 1st, the 2nd is planned to change in an upcoming version. Not sure if this answers all your questions

Aircoookie, 2020-03-18:
Yes, this is possible! Either you can use the approach @raoulteeuwen suggested with one segment per triangle. If you want to run effects like in your GIFs, you want to use the segment Grouping feature in WLED 0.9.1. Just have one segment for the entire light and set Grouping in the Segments tab to the number of LEDs per triangle! In order for WLED to retain the setting on reboot, save it to preset slot 16 and set that as boot preset in LED settings :)

AnthonMS, 2020-03-18:
Okay, the approach you are suggesting Aitcookie sounds like what I am trying to do. But just to make sure I fully understand. If I for example have a LED strip with 36 LEDs and there are 6 LEDs in each triangle the setting should look like the image below?
[image]

Thank you both for the help.

Aircoookie, 2020-03-18:
@AnthonMS yes, exactly. The 6 LEDs in each triangle will  be grouped together an always be set to the same color. Effects will render as if you only have as much LEDs as you have triangles :)

AnthonMS, 2020-03-18:
Nice, that is exactly what I am trying to accomplish. Thank you so much for the help and quick reply. You are awesome for making this project and supporting it so well. Definately going to "buy you a coffee" next month :)

closed 2020-07-24 by stale[bot]

k2OS, 2021-02-12:
Practical example of wiring DIY nanoleafs witg WLED in mind? I am planning on making a small setup myself following guides I've seen you YT.  Most if not all wire their setup so that the string goes on one long continous loop, starting in leaf #1 onto #2 and following the inside of the frames, and ending back in leaf #1 again.. in order to use WLED as far as I can tell, I need to make each leaf its own loop ('group') and have relatively long leads going on to leaf #2 and so on.. (since the end of each LED_strip will be at the beginning of each strip).

Is this how people have done it, or is there another way?

Cheers!