wled/WLED #782: NodeMCU Lua Amica Modul V2 ESP8266 ESP-12F CP2102

Cosmicbase, 2020-03-18:
Hallo sind beim NodeMCU die Pins auch

#define CLKPIN 0
#define DATAPIN 2
0 = D3
2 = D4

Connect a WS2812B-compatible RGB(W) led strip to GPIO2 (NodeMCU/Wemos pin D4). Optionally connect a normally open pushbutton to GPIO0 (NodeMCU/Wemos pin D3) and ground.

Aircoookie, 2020-03-18:
Genau, die Pinnummerierungen sind bei NodeMCU und Wemos gleich :) also stimmt 0 = D3 und
2 = D4.

Cosmicbase, 2020-03-18:
Danke. Die ist ebenfalls gleich. Für APA102 brauche ich die APA102 Binary.

Aircoookie, 2020-03-18:
Genau, in dem Fall sollte die reguläre APA102 binary ohne Probleme funktionieren.

Cosmicbase, 2020-03-18:
Wenn ich die Daten wie Port und IP Adresse in Hyperion.ng eingebe komme ich per WLAN mit dem Handy nicht mehr auf das WebGui von WLED drauf. Weder per App noch per Browser.

Cosmicbase, 2020-03-19:
Kann es sein das wenn der NodeMCU oder D1 Mini ohne Stromversorgung von ca. 10 Minuten die Daten verliert?
Ich finde weder den AP noch die IP Adresse im Netzwerk.

Cosmicbase, 2020-03-19:
Kann ich bei APA102 grundsätzlich nur die WLED_0.x.x_ESP8266_apa102.bin benutzen?
Oder geht die auch WLED_0.x.x_ESP8266.bin ?

Cosmicbase, 2020-03-20:
@Aircoookie
So hab den Fehler gefunden. Der NodeMCU hat direkten Empfang wenn ich Ihn neben den Router platziere. Scheinbar sind die 5 Meter vom TV Rack zum Router zuviel. Deswegen habe ich ihn nicht mehr im Netzwerk gefunden.
Aber warum bricht die Verbindung so ein. 
Kann man das Signal vom ESP8266 irgendwie aufdrehen. Oder ein spezieller Kanal?

Aircoookie, 2020-03-20:
Hmm 5 Meter stellen normalerweise echt kein Problem dar. Evtl. sind zu viele Metallgegenstände (Receiver und co.) in der Nähe, das könnte das Problem sein. Bei Empfangsproblemen empfehle ich sonst ein Hardware-Upgrade auf den D1 mini pro mit externer Antenne, dieser hat eine deutlich verbesserte Reichweite.

Aircoookie, 2020-03-20:
Wegen der SSID Schreibweise übrigens, genau so wie im Router eingestellt bitte. Allgemein ist es aber nicht empfohlen, SSIDs mit Leerzeichen zu haben, aber WLED sollte damit klarkommen. Die Sonderzeichen _ und - gehen immer klar

Cosmicbase, 2020-03-21:
Ich glaube das Problem liegt grundlegend an den APA102. Sobald ich sie anschließe kackt mit immer WLED ab. Obwohl ich on der Live Vorschau in Hyperion.ng alles sehe kommen die Daten irgendwie nicht sauber bei WLED an. Auch weiß ich nicht welche grundlegende Einstellungen bei WLED gemacht werden müssen so das eine Kommunikation zwischen Hyperion.ng und WLED funktioniert. 
Als Single Tool funktioniert das WLED mit APA102. Soweit bin ich schon mal.

closed 2020-07-26 by stale[bot]