wled/WLED #783: Cheap controller that WLED supports

Algram, 2020-03-18:
Hi, thanks for all the work you put into this!

I would like to ask if there are any cheap controllers available that WLED runs on out of the box?
Something like this for example: https://www.aliexpress.com/item/32890025804.html

Ideally I would like to drive about 2m (288 LEDs) with a controller and WLED.

Thanks!

srg74, 2020-03-18:
@Algram 
Look at the section https://github.com/Aircoookie/WLED/wiki/Compatible-hardware

tonyn0, 2020-03-21:
I'm not seeing anything in the wiki but development boards. He's looking for an existing LED controller that can be flashed. 

Should that be a new section or just put in Useful boards and addons?

Aircoookie, 2020-03-23:
@tonyn0 i would just add it to the `ESP8266/32 boards` section

luminxa, 2020-03-23:
Yes our new board comes with it preinstalled :) https://luminxa.com/product/luminxa-2-2-2/

[image]

huggy-d1, 2020-03-24:
I hope @Aircoookie gets a % of the profits...

luminxa, 2020-03-24:
He will :)

almulder, 2020-03-26:
Check out these. I used one to run my Addressable flood lights in my back yard. All inhad tondonwas plug into pc. Flash the wled to it and inwas up and running. They are only like $3 each. 

Amazon.com
KeeYees 3pcs ESP8266 NodeMCU CP2102 ESP-12E WiFi Internet Development Board Wireless Module Compatible with Arduino IDE https://www.amazon.com/dp/B07HF44GBT/ref=cm_sw_r_other_apa_i_4XcFEbZMDFE88

luminxa, 2020-04-06:
those are cool kind of in between the d1 and nodemcu

luminxa, 2020-04-06:
> I hope @Aircoookie gets a % of the profits...

Just an update: we haven't sold any yet, going to reach out to adafruit and see if they might be interested in distribution though.

thedannymullen, 2020-04-10:
> I would like to ask if there are any cheap controllers available that WLED runs on out of the box?
> Something like this for example: https://www.aliexpress.com/item/32890025804.html

If this box has and 8266 in it then it could be reflashed.  I would love to use one of these as it would be and out of the box led driver with no wiring needed.   I have not had the chance to purchase or research which ones have 8266 and can be reflashed though.

tonyn0, 2020-04-10:
Check the link. 

https://github.com/Aircoookie/WLED/wiki/Compatible-hardware

I added the one I know of.

thedannymullen, 2020-04-10:
@tonyn0 Thanks I missed that one!!!

srg74, 2020-04-10:
@thedannymullen In the same section there is another cheap H803 controller which can be flashed out of box too :)

closed 2020-08-08 by Algram