wled/WLED #785: Deutsche Übersetzung für's Wiki

pclin-dream, 2020-03-19:
Hallo,
ich beschäftige mich seit kurzem mit WLED und bin völlig begeistert.
Vielen dank für die tolle Software.
Ich mache Support in einigen Boards die sich mit Enigma2 Receivern beschäftigen.

WLED mit Wemos Mini D1 und nodeMCU/ESP8266 habe ich im Einsatzt.
Es läuft auf einer Drembox One Hyperion (Classic) über Wlan einwandfrei.
Auf anderen Enigma2 Receivern mit Enigmalight über USB (Adalight/Momo).
Bisher habe ich mit WLED WS2812B, SK6812-NW und LPD8066 am laufen.
Von den SK6812-NW (RGBW) bin ich völlig begeistert, ich habe noch kein besseres Ergebniss gesehen.

So wie ich die User kenne möchten diese dann natürlich auch alles in Deutsch.
Ich habe mir also das Wiki geclont und Angefangen alles zu übersetzen.

Jetzt meine Frage besteht Interesse das auch hier in Deutsch anzubieten?
Ich würde das dann gerne zur Verfügung stellen.

gruß pclin

Cosmicbase, 2020-03-20:
Hi Pclin, wir kennen uns ja schon vom Hyperion Project.
Also von meiner Seite wäre großes Interesse.
Was echt stark wäre ,eine deutsche Umgebung im WLED.
So könnte das Verständins noch viel größer werden.

StefanRu1, 2020-03-20:
Hi pclin,

ob das jetzt auf Englisch oder Deutsch ist, ist mir eigentlich egal, aber sicher nicht verkehrt.

Offtopic:
Ich habe auch auf einer DM900 Enigmalight per USB an einem Aduino Uno laufen.
Das funktioniert sehr gut.
Frage mich ob das auch per WLED machbar wäre.
Soweit ich gesehen habe kann Enigmalight ja mit Wled kommunizieren?

Vielleicht können wir das in einem Enigma2 board klären.
Auf welchem board hatte ich dich gesehen?

Gruß,
Stefan

pclin-dream, 2020-03-20:
Enigmalight geht einwandfrei mit WLED habe ich auch hier schon geschrieben.
Ich bin in diversen Boards zu finden (BoxPirates, dreambox-forum, newnigma2) um nur ein paar zu nennen.
Dort schreibe ich über WLED mit Enigmalight und Hyperion und gebe Support dafür.

Du kannst auch einfach in google 
pclin WLED enigmalight oder pclin WLED hyperion 
eingeben.

Die Übersetzung habe ich fast fertig.
Stelle Sie dann zur Verfügung.

gruß pclin

Aircoookie, 2020-03-20:
Toll dass dir WLED so gut gefällt. Gute Idee!
Ich hoffe dass ich bald eine Mehrsprachenfunktion in WLED einbauen kann damit es das komplette Interface auch auf Deutsch gibt.
Was das Wiki angeht, gute Arbeit! Am besten wir machen ein neues Repo fuer das Wiki und linken von der englischen Version darauf!

Viele Grüße

pclin-dream, 2020-03-20:
Das hört sich gut an, ich freue mich wenn ich etwas zun Projekt beitragen kann.
Ich habe die .md alle nach dem gleichen Muster umgesetzt.
Sollte Morgen fertig werden.
Zukünftigen Änderungen werde ich dann auch einpflegen.

gruß pclin

pclin-dream, 2020-03-21:
Hier die Übersetzung des Wiki

[WLED.wiki_deutsch.zip](https://github.com/Aircoookie/WLED/files/4364416/WLED.wiki_deutsch.zip)
 
gruß pclin

salopette, 2020-03-22:
> WLED mit Enigmalight und Hyperion

Hi, ich finde es auch eine sehr gute Idee deutsch mit aufzunehmen!
Für die Leute wie ich, die kein English können :-(

PS: Zu enigma boxen: was Braucht man alles dafür?
Reicht nur die Box und WLED, oder muss man noch einen Pi mit Hypriton haben?
Ich habe eine Vu Ultimo 4k, würde meine 5 WLED lampen auch gerne in die Steuerung mit der box nutzen!

pclin-dream, 2020-03-22:
Box mit enigmalight und ESP8266 mit WLED an USB.
Du kannst auch mal in der google Suche
 [Plugin] Enigmalight / Ambilight für Enigma2 Receiver (OE1.6, OE2.0 ,OE2.2, OE2.5) Dreambox, openATV, VTI
eingeben, wir haben jetzt in einem Board seit kurzem einen neuen Bereich in dem es nur um Enigmalight, Hyperion, ESP8266, WLED, Soft und Hardware. Da gibt es auch Support.

gruß pclin

salopette, 2020-03-22:
Ich kann mich da nicht registrieren.

meine hotmail.com und die gmail.com werden als ungültig angezeigt.

pclin-dream, 2020-03-22:
Ja hotmail.com geht nicht bei gmail.com muss ich nachfragen.
Bekommst gleich eine Antwort.

gruß pclin

pclin-dream, 2020-03-22:
Sind leider beide gesperrt.
Es gab zu viele Spam bots über diverse Provider.

gruß pclin

Aircoookie, 2020-03-23:
Haha ich benutze nur gmail :)
Wir haben sonst auch noch https://wled.discourse.group/ als Forum!

Habe gerade Deine Übersetzung auf ein neues Repo gehauen: https://github.com/Aircoookie/WLED-wiki-DE/wiki/
Irgendwie scheint auf manchen Seiten das Markdown kaputt gegangen zu sein und alle Links verweisen natürlich noch auf die jeweiligen englischen Seiten.
Werde das in den nächsten Tagen noch ein wenig optimieren.

Danke auf jeden Fall schon einmal und viele Grüße von mir!

pclin-dream, 2020-03-23:
Super danke dir!
Das wird einige freuen und es einfacher machen.
Die ersten Links gehen ja auch schon.

gruß pclin

pclin-dream, 2020-03-24:
Ich würde dir die Arbeit ja gerne abnehmmen die Links anzupassen, aber ich mache da wohl etwas falsch. Man kommt aber zumindestens schon auf die Deutsche Übersetzung, bei meinen Anpassungen.

gruß pclin

pclin-dream, 2020-03-24:
Ich würde dir die Arbeit ja gerne abnehmmen die Links anzupassen, aber ich mache da wohl etwas falsch. Man kommt aber zumindestens schon auf die Deutsche Übersetzung, bei meinen Anpassungen.
**EDIT:**
Ich habe es herausgefunden. 
Ich würde mich weiter darum kümmern, dann kannst du dich mit wichtigerem beschäftigen.
Auch Zukünftige Änderungen des Wiki pflege ich dann gerne ein.

gruß pclin

pclin-dream, 2020-03-24:
Bei Schnittstellen bekomme ich eine Link Anpassung nicht hin, wäre schön wenn du dir das anschauen könntest.

gruß pclin

pclin-dream, 2020-03-29:
Hallo Aircoookie,
wo finde ich die genauen Änderungen vom Wiki
Ich mache wenn ich kann zwar täglich einen git pull und sehen dann wenn Änderungen gemacht wurden und in welchem Bereich, das vergleichen was sich aber geändert hat um es dann zu übersetzen ist aber aufwendig und Fehleranfällig.

gruß pclin

MarioPeters, 2020-05-12:
Cool. Ich hab schon lange überlegt mir Ambilight nach zu bauen. Ich hab dann mit 4k leider aufgegeben, aber wenn ich den Beitrag hier lese ergibt sich folgende Frage:

Kann ich auf meine Enigma2 Box Enigmalight, oder ähnliches laufen lassen und darüber dann einen ESP ansteuern, der die LED ansteuert?
Ich wollte erst alles mit einen Raspberry Pi machen, aber für 4K ist die Variante sehr teuer und ghet glaub ich auch nicht mit HDR. Deshlab hab ich lieber auf Ambilight verzichtet, statt zukünftig nur noch in FulHD zu gucken.

pclin-dream, 2020-05-13:
Ja das geht mit einer Enigma2 Box!
Mit einer Dreambox One mit Hyperion über Wlan oder USB.
Mit anderen Enigma2 Receivern mit Enigmalight über USB mit einem VTI Image auch mit Wlan.

Mit der Dreambox One und Hyperion geht auch ein Externer Grabber je nach Hardware (Splitter, Switch, Grabber) geht dann auch 4K und HDR.

Gib im Browser **Insanelight - Hardware - Software - Plugins**  ein, dann findest du jede Menge Informationen und Software für Enigmalight.
Hyperion & Plugin HyperionControl
Enigmalight für enigma2 Receiver OE1.6 bis OE2.5
WLED Flash Tool
SEDUsetup Tool
und sehr viele nützliche Informationen und Anleitungen.

gruß pclin

MarioPeters, 2020-05-13:
Danke. Hab in Boxpirates forum geschrieben, aber ich bekomme es nicht zum laufen.
Weder mit der Version vom feed noch mit deiner.
Ich nutze das HDF Image auf den Octagon SF4008.
Bekomme keinen USB Port unter dev.
Wenn ich eine smartmouse anschließe ist ttyUSB0 da.
Muss ich nur enigma2-plugin-extensions-enigmalight_1.0+git54+b3d012c-r0.8_all.ipk installieren?
Oder noch mehr?

pclin-dream, 2020-05-13:
Bei meiner Version benötigst du auch noch die binarys.
enigmalight-bin-oatv_0.4-rc2_all.ipk

Über die Depends werden dann noch die Abhängigkeiten installiert.
Depends: enigma2, python-cheetah, libusb-1.0-0, kernel-module-cdc-acm, kernel-module-ftdi-sio, kernel-module-usbserial, kernel-module-ch341

Bei weiteren Fragen bitte dort wo du die Pakete geladen hast, hier geht es um WLED.

gruß pclin

closed 2020-09-18 by stale[bot]