wled/WLED #786: WLED Adalight kommunikation mit Ambibox funktioniert nicht !

Tihass1975, 2020-03-20:
Ein schönen guten Abend an den (die) Entwickler von WLED,

Da ich im Hyperion Forum gelesen habe das WLED die Adalight Ausgabe beherscht, wollte ich das ganze mal mit Ambibox ausprobieren. Da Ambibox keine Wlan Schnittstelle beherscht, habe ich die Kommunikation mittels USB versucht, leider ohne Erfolg ! WLED selbst, habe ich dank des erstklassigen Wiki in wenigen Minuten erfolgreich in meinem Netzwerk zum laufen bekommen. Ich verwende ein Wemos D1 mini mit APA102 LED Streifen mit der WLED_0.9.1_ESP8266_apa102.Bin .  Ich habe schon unterschiedliche USB Ports ausprobiert, es kommt keine Kommunikation zustande, ich habe stehts den richtigen COM Port gewählt. Einziges Phänomen ist, wen ich Ambibox beende, blinken alle LED`S wie nach dem Bootvorgang des Wemos. Wenn ich ein Arduino Nano mit einem Adalight Sketch bespiele funktioniert das Abilight einwandfrei ! Kann es sein das WLED das Adalight parsen nur über Wlan beherscht, oder muss ich im Sketch was anpassen und selbst kompelieren ?
Für Hilfe bin ich sehr dankbar, ich würde sehr gern den Wemos mit WLED verwenden. Und ein ganz großes Dankeschön an dieses tolle Projekt !

Gruß
Matthias

Aircoookie, 2020-03-23:
Schönen guten Abend Matthias,

genau, WLED beherrscht Adalight, allerdings auch wirklich nur über USB, also hast du soweit schon einmal alles richtig gemacht. Betreibe selbst einen ESP32 mit Ambibox. Gehen die LEDs aus während Ambibox läuft? Welche Baudrate ist in Ambibox eingestellt und wie viele LEDs benutzt du?

Ich bin froh dass du WLED magst und ich hoffe dass wir das Ambilight ans Laufen kriegen :)
Viele Grüße
Christian

Tihass1975, 2020-03-24:
Hallo Christian,
Die Leds leuchten wärend Ambibox eingeschaltet ist, aber mit den Einstellungen von WLED. Ich habe es mit Baudraten von 50.000 und 100.000 ausprobiert es findet keine Kommunikation statt, der COM Port ist richtig eingestellt. Ich habe es jetzt auch mit Hyperion probiert, mit USB gleiches Problem. Ich bekomme nur über Wlan und UDPraw eine Kommunikation mit Hyperion hin, damit funktioniert es einwandfrei ! Gibt es eine Einstellung die ich bei WLED setzten muß ? Hast du das mit der 0.9.1 unter Ambibox am laufen ?

Gruß
Matthias

Aircoookie, 2020-03-24:
Hi Matthias,
probier bitte einmal 115.200 Baudrate aus, das ist die die WLED standardmäßig benutzt. Einstellungen o. ä. brauchst du eigentlich nicht, sollte out-of-the-box laufen.
Hoffe das hilft :)

Gruß
Christian

Tihass1975, 2020-03-25:
Hall Christian,

Das war es, es läuft !!!! Allerdings ist die Performance mit Wlan und dem UDPraw Protokol deutlich schneller, liegt das an der niedrigen USB Baudrate ? Ich betreibe 224 APA102 LEDS hinter mein TV, kann man die Baudrate höher setzten würde das was bringen ? Das wäre ja auch eine Interessante Einstellung in der GUI von WLED ! Vielen Dank für dein Support !

Gruß
Matthias

closed 2020-07-30 by stale[bot]

polykoma, 2023-04-29:
I wish this would have been soolved. RIP