wled/WLED #788: Change start led

Maurotb, 2020-03-22:
Hi,
i have 894 led ws2815 , my strips are installed as an O.
To keep the length of the signal cable short, I need to connect it to a specific point, but so some effects where you need to have the start / end pixel in a specific position are bad to see.
There is a possibility to shift the first led on another position via code?
Thanks

Aircoookie, 2020-03-23:
Hi!
You can set a different starting point using segments, but that would keep the LEDs before this point off. What you really want is for the LEDs to wrap around. That is not possible right now, but easy to add! Will be an option in version 0.9.2 :)

Maurotb, 2020-03-24:
Good!
It would be nice to be able to do it on segment 0 so that we can have transactions :)

Geraldidi, 2020-08-30:
I was meesing with a rectangular shape and would love to change start led position so visually looks good. I'm using it with LedFX so I think setting it as a segment configuration would work with it

Pe-MaKer, 2020-11-05:
I found this feature request after posting on discord:
https://wled.discourse.group/t/right-placement-of-led-ends-behind-tv/1239 

I'd like to set the first pixel indepentantly from the physical start/end of the strip I placed my TV in the usual square shape. ;)
Hopefully, this FR will make it in the next release.

Aircoookie, 2021-06-30:
Implemented!

closed 2021-06-30 by Aircoookie

coryking, 2021-07-01:
Dude, nice!!!! Thanks!

rngDeadeye, 2024-11-16:
How do you do it?

DedeHai, 2024-11-16:
@rngDeadeye please use discord for support questions