wled/WLED #790: Request for Pacifica animation from fastled

emdeex, 2020-03-23:
First off... loving WLED since I found it this morning!

Second ... can I request we get the new Pacifica animation that fastled has?  Not sure yet if this is non-trivial or not.

https://www.reddit.com/r/FastLED/comments/eordjm/pacifica_new_water_animation_dedicated_to_dan/

Aircoookie, 2020-03-23:
Rest in Peace Dan.

This effect is absolutely mesmerizing, I will add it to WLED asap!

Aircoookie, 2020-03-25:
Added! However I still want to expand on it a little bit soon to be able to set different colors (although these nice blues will always stay the default!)

closed 2020-03-25 by Aircoookie

emdeex, 2020-03-25:
Fantastic!  Can't wait to try it... I'm using the WLED binary currently,
any chance of whipping a new binary version up?

NietGiftig, 2020-06-08:
@Aircoookie 
Did you have a try for expanding it with different colors, or is this not possible?
This a awesome effect !!

Aircoookie, 2020-06-08:
@NietGiftig glad you like it! Yes, Pacifica now supports palettes, so you can pick any you like most. Or select the palette  `* Color Gradient` in the name, it will allow you to set 3 custom colors :)

NietGiftig, 2020-06-08:
@Aircoookie 
Great, thanks, had not discovered the correct way to integraded the palette in the effect.
Using WLED just for a couple of hours
rtfm, as always the best solution

commit 07d6bfa9 (2020-03-25): Added Pacifica effect (closes #790)