wled/WLED #792: WLed 0.9.1 stopped working

frzme, 2020-03-23:
Yesterday I had the behaviour that my WLed controller (0.9.1) was still reachable via Wifi but the web interface reported only errors when controlling the LED strip and the HomeAssistant integration showed offline. Unfortunatly I did not record the errors that the web interface reported, sorry.
The LEDs remained off.
I could reboot the controller in the Web Interface and then everything was well again.

Any ideas what might cause this or what I could do to prevent it in the future?

Aircoookie, 2020-03-23:
Hi, sorry for the issue!
This is most likely an issue with running out of memory. How many LEDs do you have connected?
The issue might be alleviated in current master source code and the next release because I increased the free heap by around 2k, so trying that may help. If you are using the binary, just tell me your controller and LED type and I will compile a custom one for you :)

frzme, 2020-03-23:
I'm using an wemos d1 mini (esp8266) with the LED striep connected to pin 3.
137 LEDs are in the stripe.
Thank you! I'm unsure how to replicate the issue exactly as it had been running fine for a while before.

Aircoookie, 2020-03-24:
Thanks! Ledpin 3 actually has memory issues more often since the DMA driving method for that pin uses 4x as much memory as other pins. Here is a binary of the latest development version: https://drive.google.com/open?id=1Xze6wzSWQwOfjS0aiiMRrruq6Hx0-FY4 :)

frzme, 2020-03-25:
Intresting, thanks! So would you suggest that I switch to pin2? Does that have any disadvantages?

Aircoookie, 2020-03-25:
If the issues continue, yes. The only disadvantage would be that the blue onboard LED is lit permanently. You could also try a different pin like gpio5. The creator of the Neopixelbus library advises against that, but I had no major issues so far.

closed 2020-07-31 by stale[bot]