wled/WLED #795: Would like to see more sliders/controls

atuline, 2020-03-24:
I would love to see more sliders. If the html weren't compressed, we'd probably be able to figure out how to add that ourselves. The reason is that our audio reactive fork could do with a couple of extra sliders, especially the FFT version that apleschu is working on.

Even non-reactive routines could do with more, such as sine waves. . . frequency, time shift, amplitude and so on.

Aircoookie, 2020-03-24:
Hi, I agree that more sliders would be useful and at least adding a Phase slider (for having Wipe effects on different segments out of sync, for example) is something I am thinking about.
The problem is that not all effects will use every slider (even now there are effects which do not, for example, use intensity or palettes), and I believe it would be confusing to have sliders that don't do anything unless a certain effect is set. I'll think about it, maybe there is a way to dynamically display the respective sliders depending on the effect selected without causing too much overhead.

Modifying the HTML is quite straightforward despite the compression. I've made a small guide here: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality#web-ui

Absolutely love the work you guys are doing on the audio reactive fork by the way! It might be something we can merge into base at some point :)

tonyn0, 2020-03-25:
I think there are already some effects where the slider does nothing.

atuline, 2020-03-25:
There are definitely some effects where the slider does nothing, and hiding sliders for those would be a good thing. You could have dynamic sliders where, by default, Speed and Intensity are enabled (for backward compatability).  IF those sliders were not enabled by default, animations would have to be updated to enable and access them as required.

Dynamic sliders could be optionally made available where you can set for any given animation:

SLIDER1.enabled = TRUE;            // Hide/unhide a slider.
SLIDER1.name =  "Speed";          // Set the display name of a slider.
uint8_t myVal = SLIDER1.value;  // Read value as usual.

You would need to reset sliders upon choosing a new animation.

In the meantime, I'm starting to disassemble index.htm, wled19_json.ino to see if I can figure out how it all currently works (will need to learn some JSON and Javascript in the process).

blazoncek, 2022-07-12:
Partially implemented in #2610 (and follow-up commits)