wled/WLED #797: Red light chassing string on any effect

almulder, 2020-03-25:
So I am running  0.9.1 and reguardles of the effect I select the last led becomes red and jumps from the last to the next led working it's way to the first led and then it stops for about 1 min and then it starts again. Moving from the last led to the first led. I only have 11 LEDs on the strip. 

Any idea what is causing this?

Aircoookie, 2020-03-25:
Hi,
another user had this issue a few months ago, it is presumably caused by flash corruption.
You can try completely erasing the flash memory with esptool (`esptool.py erase_flash`) and then re-flashing WLED, that could fix it :)

almulder, 2020-03-25:
Aw ok, I will try that tonight when it gets dark again outside. Thanks!

almulder, 2020-03-26:
Yes this fixed the issue. Thanks.

closed 2020-03-26 by almulder