wled/WLED #798: how to flash WLED in Novostella Flood light?

Dreamoffice, 2020-03-25:
Hello 
today i got those lights
[image]
i flashed it with tuya convert with tasmota, but i m not happy how tasmota handle the ww-led, cw-led and the rgb-led.  mixing colors is only possible with mqtt commands. in WLED it works much much better. is there a chance to flash WLED to the  novostella flood lights?
the configuration looks like this
[image]
Br Dave

fishbone-git, 2020-03-25:
From config it looks like one output per LED.  If tasmota can go on there, so can WLED but the problem is that WLED can currently only do one output.  There is work being done to add support, but that will be some time.  If you just want to see if there will be a future worth waiting for, you should be able to upload a wled binary onto tasmota_mininal and see what you get for control of one LED.  You would need a custom build as WLED release with either GPIO2 or 3 for output.

personally, I would probably look at ESPhome for a device like this one.  I did the same for an RGB bulb (tuya_convert->tasmota).  Same bad colour transitions, but ESPhome is better and you do have some ability to configure basic single LED effects. Plus, it will do multiple PWM outputs

Dreamoffice, 2020-03-25:
thanks for your reply. 
do you have an idea how to flash esphome to the device.? i don t have access to the board, only via wifi

fishbone-git, 2020-03-25:
You can compile esphome without uploading it, then select that binary when updating from Tasmota.  You may need to load the tasmota_minimal binary first to have enough room for esphome.  If you know the IP address, you can also edit the esphome config file to use that IP address and it should upload when it builds. It is a bit more advanced that the tuya-convert process but once you have it set up, it's a fairly simple web page so manage the esphome config.  You will likely find some good examples to borrow which will get you at least part of the way

huggy-d1, 2020-03-25:
Have you looked at https://github.com/Aircoookie/WLED/blob/master/wled00/NpbWrapper.h and editing it for analog LEDs and  your configuration?  WLED will control analog strips too!

Dreamoffice, 2020-03-25:
@huggy-d1  thanks will try this tomorrow. Can in Upload wled to the board of I flash tasmota minimal on it?

Dreamoffice, 2020-03-30:
how can i flash WLED to the device? 
i just made few changes like on the screenshots below:
[image]
[image]
is this all i need to change?

Dreamoffice, 2020-04-05:
Hi 
just a short Summary to flash WLED on the floodlights. 

1. Flash the Floodlight with Tasmota using Tuya-Convert
2. Flash Tasmonta-minimal.bin via the Tasmota UI
3. Go to the Tasmota Console and enter the Command `Setoption78 1`
3. Compile WLED with the 5Ch PWM LED configuration. (check code below)
4. Flash the WLED Firmware.bin File via the Tasmota UI to Floodlight
5. Configure WLED...... have fun

BUT i have a problem with the CW-LED. i can't control them. does anybody has an idea what i do wrong. 
I changed the Pins of the analog LED in the File NpbWrapper.h. Maybe i made some mistakes there???

`//this code is a modified version of https://github.com/Makuna/NeoPixelBus/issues/103
#ifndef NpbWrapper_h
#define NpbWrapper_h

//PIN CONFIGURATION
#ifndef LEDPIN
#define LEDPIN 2  //strip pin. Any for ESP32, gpio2 or 3 is recommended for ESP8266 (gpio2/3 are labeled D4/RX on NodeMCU and Wemos)
#endif
//#define USE_APA102  // Uncomment for using APA102 LEDs.
//#define USE_WS2801  // Uncomment for using WS2801 LEDs (make sure you have NeoPixelBus v2.5.6 or newer)
//#define USE_LPD8806 // Uncomment for using LPD8806
//#define USE_P9813   // Uncomment for using P9813 LEDs (make sure you have NeoPixelBus v2.5.8 or newer)
#define WLED_USE_ANALOG_LEDS //Uncomment for using "dumb" PWM controlled LEDs (see pins below, default R: gpio5, G: 12, B: 15, W: 13)
#define WLED_USE_H801 //H801 controller. Please uncomment #define WLED_USE_ANALOG_LEDS as well
#define WLED_USE_5CH_LEDS  //5 Channel H801 for cold and warm white

#ifndef BTNPIN
#define BTNPIN  0  //button pin. Needs to have pullup (gpio0 recommended)
#endif

#ifndef IR_PIN
#define IR_PIN  4  //infrared pin (-1 to disable)  MagicHome: 4, H801 Wifi: 0
#endif

#ifndef RLYPIN
#define RLYPIN 12  //pin for relay, will be set HIGH if LEDs are on (-1 to disable). Also usable for standby leds, triggers,...
#endif

#ifndef AUXPIN
#define AUXPIN -1  //debug auxiliary output pin (-1 to disable)
#endif

#ifndef RLYMDE
#define RLYMDE  1  //mode for relay, 0: LOW if LEDs are on 1: HIGH if LEDs are on
#endif

//END CONFIGURATION

#if defined(USE_APA102) || defined(USE_WS2801) || defined(USE_LPD8806) || defined(USE_P9813)
 #define CLKPIN 0
 #define DATAPIN 2
 #if BTNPIN == CLKPIN || BTNPIN == DATAPIN
  #undef BTNPIN   // Deactivate button pin if it conflicts with one of the APA102 pins.
 #endif
#endif

#ifdef WLED_USE_ANALOG_LEDS
  //PWM pins - PINs 15,13,12,14 (W2 = 04)are used with H801 Wifi LED Controller
  #ifdef WLED_USE_H801
    #define RPIN 04   //R pin for analog LED strip   
    #define GPIN 12   //G pin for analog LED strip
    #define BPIN 14   //B pin for analog LED strip
    #define WPIN 13   //W pin for analog LED strip 
    #define W2PIN 05  //W2 pin for analog LED strip
    #undef BTNPIN
    #undef IR_PIN
    #define IR_PIN  0 //infrared pin (-1 to disable)  MagicHome: 4, H801 Wifi: 0
[... 227 lines cut ...]
  {
    switch (_type) {
      case NeoPixelType_Grb:  return _pGrb->Pixels();  break;
      case NeoPixelType_Grbw: return _pGrbw->Pixels(); break;
    }
    return 0;
  }

private:
  NeoPixelType _type;

  // have a member for every possible type
  NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>*  _pGrb;
  NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>* _pGrbw;

  void cleanup()
  {
    switch (_type) {
      case NeoPixelType_Grb:  delete _pGrb ; _pGrb  = NULL; break;
      case NeoPixelType_Grbw: delete _pGrbw; _pGrbw = NULL; break;
    }
  }
};
#endif
`

fishbone-git, 2020-04-05:
>    4. Compile WLED with the 5Ch PWM LED configuration. (check code below)
>    5. Flash the WLED Firmware.bin File via the Tasmota UI to Floodlight
>    6. Configure WLED...... have fun
> `//this code is a modified version of Makuna/NeoPixelBus#103

Have you set the led settings in the UI to 5-channel?

Dreamoffice, 2020-04-05:
@fishbone-git where can i change to 5Ch in the UI? i only see 4ch :-(
[image]

fishbone-git, 2020-04-05:
Apologies.  I meant the 4, but typed 5 from reading your mention of 5ch.  I'm not too familiar with the novostella flood.  It's RGBWW/CW?  Rather hard to read the code you've pasted but it looks like you have the 5CH defined.  Any chance the pin is wrong?  Perhaps someone else has tried the same light (I've seen them mentioned a few places on the hobby sites) and they'll have some input.

Dreamoffice, 2020-04-05:
Which hobby site so you mean?

Kosh42, 2020-04-14:
WLED is really geared up for digital strips. I used it on analogue, but soon went back to ESPHome.

lloydpick, 2020-10-11:
I'm also looking to put WLED onto these floods, mainly because I'd like E1.31 support for them.

If anyone has anything they can share I'm all ears.

jarjar34, 2020-10-13:
Purchased the same floods and love them but never knew they had E1.31!!! How did you all get WLED on them and did you use the code above? I'm a bit of a rookie at all this, but I am actively used WLED and love to integrate this into my xLights.

Kosh42, 2020-10-13:
Only way I've seen e1.31 is here, but not tried it myself: https://github.com/thehookup/E131-MQTT-RGBW-Lights

Shame the ESPHome e1.31 support is addressable only too.

huggy-d1, 2020-10-13:
> ...
> I've got WLED onto the hardware, web UI all working, had to use the 1mb version to get it to fit. I can turn the light on and off, but I can't change the color.

Did you follow the instructions above by @Dreamoffice from Apr 5?  They stated they used the 4CH setup, with the 4 PINs configured same as TASMOTA.

lloydpick, 2020-10-13:
> Did you follow the instructions above by @Dreamoffice from Apr 5? They stated they used the 4CH setup, with the 4 PINs configured same as TASMOTA.

Not quiet yet, I need to get sometime to reformat the code that was posted above and then compile myself. I was just using the pre-compiled binaries as a first step.

lloydpick, 2020-10-13:
> Only way I've seen e1.31 is here, but not tried it myself: https://github.com/thehookup/E131-MQTT-RGBW-Lights
> 
> Shame the ESPHome e1.31 support is addressable only too.

Ah, I hadn't found that repo. If I can't get WLED working, then I may fall back to that.

jschwalbe, 2020-11-30:
Anyone get progress with these? If so, would you mind posting a bit about how? 
I found this repo: https://github.com/hhromic/e131-mqtt-bridge but haven't had luck getting it to work.

solarkennedy, 2022-03-13:
I was able to get these to work on my Lamps.
No software changes are required on a modern (0.13) version of WLED!
Getting WLED on them in the first place is tough though.
https://www.xkyle.com/WLED-on-a-Novostella-Flood-Lamp/

lloydpick, 2022-06-02:
For me I finally have them under control of Xlights, however its not using WLED (although it could). 

I had orignally flashed my floods with standard Tasmota so that I could have some basic control through HomeAssistant with scenes. However in version 10.1.0.4 of Tasmota they added a tiny line to the change log... `DDP schemes for light and WS2812 (#14017)`. But this isn't enabled by default, so I used a custom compiled version of Tasmota found [here](http://tasmota.cloudfree.io/tasmota-ddp.bin.gz) with it enabled. 

Once flashed run the console command `Scheme 5` and that will enable DDP. You can then either point WLED at the Tasmotas with...

[image]

Or you can point XLights directly at the floods...

[image]

I have the FPP proxy configured since my show runs from a dedicated FPP controller, rather than the computer running XLights.

PKCubed, 2023-11-09:
How do I take the front glass off without breaking it?