wled/WLED #801: Request. - auto on / off

almulder, 2020-03-26:
Can we get a feature to auto turn on and auto turn off based on time. Like add a start time and end time. 

Even better to be able to select sunset to sunrise with a +/- in mins option. Say 30 min before sunset and 30 min after sunrise?

I am sure there are ways to do it with third party stuff. But it would be nice to have it built in.

srg74, 2020-03-26:
Please read Wiki https://github.com/Aircoookie/WLED/wiki/Macros

almulder, 2020-04-12:
I looked but it does not explain stuff clearly. 

Can someone post a macro to turn on lights at 6:00pm and leave on for 6 hours or turn off at 12am.

raoulteeuwen, 2020-04-13:
Not sure since i don't use them, but afaik it should work like so: go to Settings/Time&Macros. I assume you need to turn on NTP at the top (else WLED has no idea about the time). At the bottom, you can specify macro's to start at a specific time. If all you want is turn WLED on at time X and turn it off at time Y, program macro 1 with code "T=1" and macro 2 with code "T=0" (both without the quotes). Than put macro 1 in time slot 1 (with time X) and macro 2 in time slot 2 (with time Y) .

This is basically what is also described at https://github.com/Aircoookie/WLED/issues/101#issuecomment-457757212 .

But when i tried with my own WLED before posting this, i could not get it to work. But please try ;-).

franzuu, 2020-04-16:
I stumbled upon this library:
https://github.com/dmkishi/Dusk2Dawn
It would be nice if this could be implemented.

DJL-F3D, 2020-05-28:
I completely agree I don't know how to add to the wiki but this is what I use (Gleaned from Dr Zzzzz and other sources)

First you have to define the macro's, these are mine so drop them each numbered spot (5 total).
Macros:

T=0
T=1
PL=~
T=2
ND

This is what they do - Can't find any good info in the docs about this stuff.

(M1) T=0 (power off)
(M2) T=1 (power on)
(M3) T=2 (power toggle) assign to long press

(M4) PL=~ (next preset) assign to short press macro

(M5) ND = Shutdown timer activate assign to double press

I have them assigned to these button pushes:

Short press macro = 4 next preset
Long press = 3 power on/off
Double press = 5 shutdown timer activate

Then at the bottom of the 'Time & Macros' screen you can setup your time things come on and off. So in the morning my turn on macro Macro 2 - you just need 2 in the macro spot. triggers on weekdays at 05:30 

[image]

[image]

raoulteeuwen, 2020-05-29:
Hi @DJL-F3D did you find https://github.com/Aircoookie/WLED/wiki/Macros (and the link on that page to https://github.com/Aircoookie/WLED/wiki/HTTP-request-API )? Or that isn't clear enough? Feel free to improve it!

DJL-F3D, 2020-05-29:
wow seems aggressive. But for a non programmer like me I find it hard to parse that kind of information without examples being shown. Sorry to bother you

DJL-F3D, 2020-05-29:
Thanks for the reply. For a non programmer like me, I find it hard to apply that sort of list without some working examples. Aircoookie has all the functionality in there. It’s just hard for someone like me to apply it.

Legsmaniac, 2020-05-29:
Perhaps an easier (and cheating?) way would be to use a timer switch?  Plug the power supply into the timer switch and ensure WLED is set to always start ON in the settings.  You could even consider one of the many available plug-in WiFi switches which can also be programmed with schedules.  
Just a thought.  No macros needed.  😉

DJL-F3D, 2020-05-30:
It really is worth persevering with the functionality already in WLED though. 
In the morning my phone alarm goes off and exactly the same time my WLED lights com on too because of the internet time syncing. Good thing is, you’ll never have to set the time yourself. Just trigger the macro to switch on at the same time.

It’s really easy once you give it a crack. From what i’ve read, WLED is really designed to be triggered externally by other home automation setups, So this being built in is really a bonus for those not wanting to go down that route.

garyd9, 2020-08-09:
for the sunrise/sunset functionality, there is now a usermod:   https://github.com/Aircoookie/WLED/tree/master/usermods/UserModv2_SunRiseAndSet

closed 2020-12-16 by stale[bot]