wled/WLED #802: Compilinmg WLED for default_envs = esp8285_5CH_H801 throws error

Dreamoffice, 2020-03-26:
Hello
i m trying to compile WLED to use it on the Novostella Floodlights. These are 5 channel pwm LED lights.
therefore i did the following:

[image]
[image]

but during compiling i get this error:
[image]

`wled00\FX_fcn.cpp: In member function 'void WS2812FX::setRgbwPwm()':
wled00\FX_fcn.cpp:829:20: error: cannot convert 'RgbwColor' to 'uint32_t {aka unsigned int}' in assignment
   _analogLastColor = color;
                    ^
*** [.pio\build\esp8285_5CH_H801\src\FX_fcn.cpp.o] Error 1`
[image]
 
i didn't get the problem....

br Dave

huggy-d1, 2020-03-26:
I proposed an edit to FX.h that changes _analogLastColor to datatype RgbwColor.  Hope that works. Otherwise it may need to be defined locally as a class property, which is kinda weird (to me at least) for conditionally used variables and logic.

Aircoookie, 2020-03-26:
@huggy-d1 i did just that, change the type to RgbwColor yesterday. Just didn't push yet 😅 
Fixed in master!

closed 2020-10-12 by Aircoookie