wled/WLED #803: Feature Request: Show colors in pallettes

jezmck, 2020-03-26:


jezmck, 2020-03-26:
Perhaps like this?

https://codepen.io/jezmck/details/KKpromE

[image]

Aircoookie, 2020-03-26:
Not a bad idea! Once the palettes are stored in a `palettes.json` file instead of hard-coded, the browser could read this file and render previews for the palettes! (would be changed from select dropdown to buttons)

tihoangyeudau, 2020-03-27:
> 
> 
> Not a bad idea! Once the palettes are stored in a `palettes.json` file instead of hard-coded, the browser could read this file and render previews for the palettes! (would be changed from select dropdown to buttons)

i like it. very easy to choose

jezmck, 2020-03-27:
Something clever will have to happen for the special palettes I guess.

Brausmith, 2020-05-15:
I like this, and also for the favourites.

jezmck, 2021-04-03:
Awesome job in the latest release, thanks!

closed 2021-04-03 by jezmck