wled/WLED #807: Default color/pattern after powering on?

Boby71, 2020-03-27:
I played around for a while now, but always when I power on my nodeMCU the pattern is solid and the color orange by default. I's like to have solid blue when powering on the device. Is there somewhere a setting I didn't find where I can change the default color when starting up the board? Thank you!

Firmware is 0.9.1

raoulteeuwen, 2020-03-27:
Did you try:

- set color you want
- go to Favorites (save color by checking the Saving Mode and clicking a memory slot) . Uncheck Saving mode
- go to Config/LED Preferences and enter the memory slot nr of your color at "Apply preset at boot (0 uses defaults) ", click Save
- go to Config/Security Updates, scroll down and click 'Save & Reboot'

Boby71, 2020-03-27:
Worked like a charm, thank you very much!

closed 2020-03-28 by Boby71