wled/WLED #810: Add instructions to the Wiki for setting Primary/Secondary colours?

demosthien, 2020-03-28:
As the title says...

I've spent ages trying to find out how to do this and I cannot find the info anywhere. It's probably a super simple thing to do but I cannot for the life of me work it out.

Currently I'm on the Discord trying to get an answer there.

raoulteeuwen, 2020-03-28:
Hi. What version are you running? If it is the latest, on the home screen, just below the color wheel, you see a 1, a 2 and a 3 . Those stand for primary, secondary and tertiary. Click one of those and select a color. Want to save your choice? Go to favorites, check the Safe Mode, click a memory location (for example the 1st), uncheck safe mode > ready?

demosthien, 2020-04-02:
> Hi. What version are you running? If it is the latest, on the home screen, just below the color wheel, you see a 1, a 2 and a 3 . Those stand for primary, secondary and tertiary. Click one of those [...]

Thanks Raoul. I ended up finding the information in the end, but it took a very long time. Far longer than such a simple action should have. As suggested in my title, I think this information would be useful in the Wiki. probably on the same page as the list of effects and palettes given how it impacts them particularly.

closed 2020-04-02 by demosthien