wled/WLED #812: Blink on startup,only with relay

Maurotb, 2020-03-28:
Hi,
i have setup my wemos d1 mini pro ,led I use ws2815b,with relay to enable my power supply when i power on wled.
My led blink on start instead make a soft transition...

https://www.youtube.com/watch?v=jGPSlDlD0Hg

If i remove relay and make led always powered, wled work correctly with soft start of leds.
I think it's a hardware problem, the wled signal arrives before the voltage on the LEDs is sufficient i think...
How i can resolve this?
Thanks

Aircoookie, 2020-03-29:
Hi,
currently WLED turns the relay on at the same time as it turns on the LEDs (sends data). Having the relay turn on earlier would mean delaying the data signal. I might be able to add support for that in the future, but that is not ideal... What happens if you select a longer transition time in LED settings? (maybe 2 seconds)

Maurotb, 2020-03-29:
Hi, if i select 2 seconds of transition, problem persist...
I have tried to start power by shelly 200 300ms before send data.
This work.

I think the best solution is to give the option in the options to select how long to activate the power before wle and how long to turn it off after wled. 
I believe time is dependent on power supply.
I'ts possible?
I was hoping I could get rid of shelly with wled, but with this problem I can't do it
Thanks

Aircoookie, 2020-03-29:
The problem is that the transition and output to LEDs is started immediately once the power-on command is received. Changing that would not be trivial, so while it sounds simple, just turning on the relay immediately, but delaying data would need quite a few adjustments to the logic.
How long does it approx. take for the LEDs to turn on after you press the power on button?

Maurotb, 2020-03-29:
> How long does it approx. take for the LEDs to turn on after you press the power on button?
I did not understand the question well, the LEDs turn on immediately, but they make an annoying flash at power on  with random colors...
I use wled to light the house as you can see from the video, it's not nice to see that flash every time I turn it on...

I guess this problem exists in general for those who use the relay function,acceptable for effect light but not for lighting... 
Do you think it can be added as a feature in the new release?
Otherwise I build the system using shelly and making a delay via node-red but I'd like to simplify it using the wled relay output ...

huggy-d1, 2020-03-29:
What size capacitor do you have on the input to the esp and what size capacitor at the power output to the led strip? If the answer is none to either question, try adding some capacitance. You can search online for appropriate sizes if you do not know.

Maurotb, 2020-03-30:
I think my setup is okay,
i use Mean Well rsp-200-12 power supply for led.
6  2.5mm cables start from this power supply and arrive at 3 different points of the LEDs (the strips are quite long,approx 15 18mt)
At the led input there is a 1000uf capacitor

separate 5v power supply for d1 mini pro.
In d1 mini pro i have a 470uF at input and a sn74ah level shifter.

I have try to remove capacitor and doubled it,i try to remove resistor of signal at input, same problem :(
My led is ws2815b

huggy-d1, 2020-03-30:
Thanks for explaining your setup and problem in more detail.  I have to assume you are using a relay with physical contacts rather than an SSR or other fast switching relay.  The time required for physical movement of relay contacts is often in the range of the problem you describe (up to 200ms).  If your default boot effect is SOLID and color BLACK, does this resolve the problem of flashing at startup?  If you happen to have a faster switching relay available, please try that.

Maurotb, 2020-03-30:
Yes, with solid black work with no flash.
If I turn up the brightness even very little, the problem occurs

I have try with a normal CE relay finder 5v 10A connected to wled and shelly relay connected to node-red.
My electrical system is certified with CE marked components, I would like to avoid placing non-certified components on the 220v, I do not know there are 5v solid state relays in gewiss bticino or other manufactured...

huggy-d1, 2020-03-30:
Oh my, I thought you were switching 5V power, not 220VAC mains power.
There will be an inherent lag from the time the 220VAC is switched on before acceptable 5VDC is available at the LED strip.  If the 5VDC is switched, the time delay would be much less.

Maurotb, 2020-03-30:
No i switch main power supply.
I can't make a switch on low voltage (i have 2815b,12V), too much current 
and also turning off the power supply is more efficient.
Keep in mind that I don't have a single strip, but about 40mt of LEDs will be between the entrance and the corridor powered with two Mean Well rsp-200-12 (400W).
I believe that in real use, implementing the possibility of deciding the relay activation and deactivation time is important ...

austwhite, 2020-03-30:
The relay in WLED is designed with switching the stable 5V power in and out for the LEDs as far as I know.
If you switch mains power, it takes time for everything to stabilise and you will get very weird behaviour.
For switching mains power, you'd need to engage the mains power, wait for upto 1 second then engage the WLED power line separately as far as I can work out.

Maurotb, 2020-03-31:
approxx 30A.

In the datasheets of the led power supplies it is always specified to interrupt the power supply, connect the load instantly I think it can give problems on the regulation line which could give wrong voltages to the leds as well as needing very powerful relays.
See this:
https://www.meanwell-web.com/content/files/pdfs/productPdfs/MW/HLG-320H/HLG-320H-spec.pdf

However I understand that the relay pin is not server for my purpose, do you think it could be a feature that will be added?

In 600W version of power supply (150€ argh!), 
https://www.meanwell-web.com/content/files/pdfs/productPdfs/MW/HLG-600H/HLG-600H-spec.pdf
there is a power on input (5v) with hold up time of 15ms ,I think it's the only possibility right now, given the cost, have you ever tried?

Maurotb, 2020-04-12:
@austwhite 
I bought mean-well HTPG-200-12 , this power supply has a digital power on/ power off input and an hold up time of 16ms.
I connected the pin d6 via a three-state buffer to this input , now I can turn the power supply on  and off in 16ms.
The problem of the blink at power on has reduced a lot but it is not yet perfect, we can see strange colors for a moment at power on.
Don't you think to insert this delay in interface?
Could you tell me in case you want to recompile I wled with this delay where can I put it?
Thank you

Aircoookie, 2020-04-13:
@Maurotb you could try adding this to the top of `colorUpdated()`
```cpp
if (bri && !briIT) {
    digitalWrite(RLYPIN, RLYMDE);
    yield();
  }
```
this could toggle on the relay slightly earlier, maybe it helps.

colorUpdated is in `led.cpp` in current source, this corresponds to `wled08_led.ino` in older source versions.

Maurotb, 2020-04-14:
I recompiled adding the code but a blink appear on led and in webpage i see connection to light failed. please refresh page...

Aircoookie, 2020-04-14:
Sorry, my mistake, forgot about the system context yield issue. Try leaving that `yield();` out, then it should work (but might not make a huge difference)

Maurotb, 2020-04-15:
@Aircoookie ok, i removed yield(); and now work, but the blink problem on start persist :(
i think time is too short

Aircoookie, 2020-04-15:
Hmm I figured the delay might be too short to make a difference. So we might need a timer, but I don't know if even that would help. Please do a short test for me by commenting out `strip.service()` in the main `loop()` (wled.cpp in new source, wled00.ino in old). You will not get any output to the LEDs, but the webserver and relay should work and you should not get any flashing. If you'd still get flashes, there would be nothing we can do in software, but if not we might be able to delay turning on by 200ms or so.

Maurotb, 2020-04-16:
@Aircoookie ok i have try.
I commented on the strip.service () function , relay go on and there is not blink in my strip led,all led stay off.

Aircoookie, 2020-04-16:
Perfect! This means we can likely solve the problem.
Just make the following changes (I recommend updating to latest master first)

In wled.h someplace after line 150 add:
```cpp
WLED_GLOBAL unsigned long turnedOnTime _INIT(0);
```

In wled.cpp replace line 85 `if (!offMode)` with:
```cpp
if (!offMode && millis() - turnedOnTime > 150)
```

Finally, in led.cpp, add to the top of colorUpdated()
```cpp
if (bri && !briIT) {
  digitalWrite(RLYPIN, RLYMDE);
  turnedOnTime = millis();
}
```

I hope it will help you :)

Maurotb, 2020-04-19:
Good news! @Aircoookie  now is perfect!
30ms is enough with this power supply but I put 200ms for safety. 
Do you think you can insert this feature in future releases?
Thank you!

Maurotb, 2020-06-08:
@Aircoookie is possible to insert this patch in a future release?
Thanks

Aircoookie, 2020-06-08:
@Maurotb yes, I will add this as an optional patch soon since other users have wished for this as well :)
Tracking progress in #346

closed 2020-06-08 by Aircoookie

Sangoku, 2021-12-19:
> Perfect! This means we can likely solve the problem. Just make the following changes (I recommend updating to latest master first)
> 
> In wled.h someplace after line 150 add:
> 
> ```c++
> WLED_GLOBAL unsigned long turnedOnTime _INIT(0);
> ```
> 
> In wled.cpp replace line 85 `if (!offMode)` with:
> 
> ```c++
> if (!offMode && millis() - turnedOnTime > 150)
> ```
> 
> Finally, in led.cpp, add to the top of colorUpdated()
> 
> ```c++
> if (bri && !briIT) {
>   digitalWrite(RLYPIN, RLYMDE);
>   turnedOnTime = millis();
> }
> ```
> 
> I hope it will help you :)

The last part is not available. i cant compile with it.. and it is the main key as i see it. 

Also if (!offMode) is no longer if (!offMode) but if (!offMode  || strip.isOffRefreshRequred)

DieNummerEins, 2023-10-05:
@Aircoookie 
Hi Christian. Is this feature (delay for led at power up) somewhere in progress?

I have the same problem with blinking at startup. I use this SSR:
https://a.aliexpress.com/_mP11K0O

I need a real solution because I would like to install WLED everywhere in the house (every room, garage, roof edge as permanent holiday light, garden...). A preset/playlist-solution like mentioned in #346 isn't convenient if you have 30-40 WLED-controller and you want link them together (some of them should together behave as one).

I would really appreciate this feature!

blazoncek, 2023-10-05:
It has been implemented in 2020.

DieNummerEins, 2023-10-05:
Are You sure? At the end of #346 there is said, that it is still not implemented.

How can I set up the delay (if it is implemented)?

doronazl, 2023-12-11:
@Aircoookie mind directing me to how to get this solved ? having the same issue with a ssr