wled/WLED #813: Lights go offline - can still ping

JayRabxx, 2020-03-29:
I've been going crazy trying to figure this out. After some amount of time, my light string will just disconnect and go offline. Except it will be reachable (ping IP or .local url) but can't interact with or load the web portal. Shows as unavailable in Home Assistant, doesn't react to MQTT messages, nothing until I unplug the whole thing and reset it. 

It's running the latest release on a D1 mini. One strand of 230 2811's, 60A power supply, connected to a WiFi repeater I have in the garage, about 7 feet between the two.

huggy-d1, 2020-03-29:
Do you have the led strip on a vlan?  If not, try moving it to one so all the house and neighbor wifi traffic does not reach your esp.

theoracle09, 2020-04-01:
> I've been going crazy trying to figure this out. After some amount of time, my light string will just disconnect and go offline. Except it will be reachable (ping IP or .local url) but can't [...]

The D1 Mini ESP8266 version is 3v3 I believe. Are you using a level shifter from 3v3 to 5v? I don't own any 8211's, but a quick search shows them operating at 5v. I've driven 66 ws2812's with a plain old D1 mini, without level shifting, and could only push around 20% brightness. Any higher, for any amount of time, would result in the D1 mini resetting itself and getting stuck in a loop trying to provide the current the strip is asking for. 

Also, are you powering them through the D1 mini? With a 230 strip, you'd want them powered in parallel to the Mini, through a level shifter.

Aircoookie, 2020-04-01:
@JayRabxx hmm that's not good. I suspect the module is running out of memory due to simultaneous connections... You could test if disabling MQTT helps, but of course that is not a solution if you rely on it.

JayRabxx, 2020-04-01:
> Do you have the led strip on a vlan? If not, try moving it to one so all the house and neighbor wifi traffic does not reach your esp.

My network equipment is basic enough that it doesn't allow configuring vlans. 

> The D1 Mini ESP8266 version is 3v3 I believe. Are you using a level shifter from 3v3 to 5v? I don't own any 8211's, but a quick search shows them operating at 5v. I've driven 66 ws2812's with a [...]

I have the 230 strip powered by the 60A PSU, with a common ground to the D1 mini. I have a somewhat long length to the first pixel, but have never had issues with the signal being bad or anything. I guess I could try a sacrificial pixel close to the D1 to boost the signal? Not sure how a weak signal would kill the D1 though. 

@Aircoookie I don't rely on MQTT, I was only using it within Home Assistant for a panel of preset buttons. The integration is missing that, but not a deal breaker in the end. I've disabled MQTT for now and will see if I get different results.

arvind-balaji, 2020-04-27:
Experiencing the same issue. WLED goes offline after a few hours, pinging it still get's a response.
Any ideas on what I could do to try and debug this?

JayRabxx, 2020-05-03:
I never found a solution. I tried leaving the lights always on. I tried turning off wi-fi sleep mode. I tried adding a null pixel. Nothing has solved my issues and I'm not even sure how to debug it further

donseelim, 2020-05-09:
Happen to me, update OTA to 10.0 and wifi keep dropping, took the D1 mini out and reflashed it using esptool and no problem since. Strange.

closed 2020-09-13 by stale[bot]