wled/WLED #815: WLED and FPP

CaffeinatedFunctionality, 2020-03-30:
I used WLED before December and really loved it but had some bad delays with E1.31 and Xlights. When I moved to Espixelstick everything worked flawlessly. I am now moving back to WLED because, well you have done an amazing job with this product and I use it for multiple projects inside my house and I've heard E1.31 has improved quite a bit. The problem is I can't get the LEDs to work with E1.31. Same settings as espixelstick, though I can't figure out where to put universe size in the app. Espixelstick requires universe size to be placed in their UI. I don't know if that's it or not. I imagine you may already just have it default at 510 rather than the 512. Anyways, I really want to get this working for some quarantine lights for the neighborhood. I want to stick with WLED because switching from Xlights to a random sequence that you built is just awesome. I am using unicast. 450 max LEDs per controller. I'm assuming the packets are being rejected or something. I'm not sure. I need to setup Wireshark. I don't believe the controllers are in AP mode since they are connected to the network. I don't think they can be on both. Oh when using xlights it works and shows it using E1.31. When using FPP it doesn't work at all with unicast. Help. Thank you

CaffeinatedFunctionality, 2020-03-30:
I should also state that using FPP with espixelstick seems to output correctly. I'm assuming maybe a setting is incorrect somewhere in WLED. I could be wrong

CaffeinatedFunctionality, 2020-03-30:
Looks like I figured it out. FPP had my source interface as eth0 and it should be wlan0. It's weird though because it works regardless of the source interface with espixelstick. I will do some tests tonight. I think there is more to it. I think the colors are off as well. This may not have anything to do with FPP. @Aircoookie can you confirm default universe size is 510 or do I need to change it from 512 somewhere?

Aircoookie, 2020-03-30:
Hi, I can't reply to you in detail right now, sorry. But I can confirm that the correct universe size should be 510. Are all colors off or only part of the LEDs?

CaffeinatedFunctionality, 2020-03-31:
It looks like part of the LEDs. I will take a look and see if I am missing
something else on my end today

wvpt, 2020-07-13:
I am having same issue. Did you get wled to work with xlights?

closed 2020-11-19 by stale[bot]