wled/WLED #816: Report graphical result - segment’s percentage

sommo, 2020-03-31:
Hi,
The purpose of that idea is to represent data via WLED.
In my case a temperature and a ping result.
I want to split the strip into segments for example 2 segments.
A segment represent a temperature sensor (from led 1 to 100 and the other one for a Ping result, from 101 to 200, I do a ping to 8.8.8.8 and show me how fast is gone).
I have a value of A water temperature sensor that could go from 10 to 60 degrees, so for the first segment I need that if the temperature is 50, it turn on 83% of the first segment in red colour. I could do it via rest API but I down know how to turn on only a piece of a segment.
As the same way I will manage the second segment, always via a bash script, that ping 8.8.8.8 and turn on an amount of led in the second segment that rapresent the percentage a ping from 0 to 100ms for example.
So the question is, how to turn on a % of segment, is there an api call that can do that?or I need to create as much segment as I need to represent some stage of the percentage?
Thank you!

raoulteeuwen, 2020-03-31:
Hi @sommo . You know there is an effect called Percent , which you could set per segment? You can try http://<your WLED IP>/win&IX=100 and play with the nr. https://github.com/Aircoookie/WLED/wiki/HTTP-request-API shows how to set it for a segment.

sommo, 2020-03-31:
> Hi @sommo . You know there is an effect called Percent , which you could set per segment? You can try http:///win&IX=100 and play with the nr. https://github.com/Aircoookie/WLED/wiki/HTTP-request-API shows how to set it for a segment.

Seems to work like I want! Thank you!!

Inviato da iPhone

closed 2020-04-01 by Aircoookie