wled/WLED #819: Send JSON commands via MQTT

grumpazoid, 2020-04-01:
I am trying to figure out the API so I can use in node red.
I wish to get current state, do something, and then write that initial state back all using MQTT.

So far I can get the current state by 
1) sending a /json/state http request - response is a JSON object 
2) subscribing to wled/1/v  - this gives me the information as XML

Is it possible to receive the current state in JSON format by sending a MQTT message?

Also is it possible to send that JSON object back over MQTT  to set the state. The documentation seems to say that I can only send http commands over MQTT.

Aircoookie, 2020-04-01:
Hi!
The JSON API is not currently supported via MQTT, only over HTTP.
This has been requested many times though, so I will make it possible soon :)
Here is the issue for tracking it: #249

closed 2020-04-01 by Aircoookie

grumpazoid, 2020-04-02:
@Aircoookie 
Awesome.  Thank you so much for this amazing project.