wled/WLED #822: ESP8266 Getting Offline all the time EVERY 10 seconds and not connected back till power off

aryedevops, 2020-04-03:
Hi,
New firmware makes it getting Offline all the time. it works for 10 seconds and then offline and need to get out of the wall and back to work for ONLY 10 seconds more. 
Same board works great with 0.86 version for a while and no issues - just wanted to upgrade and with no success at all.
Board is standard WiFi ESP8266MOD data connected on D4.
Please let me know what to do in order to make it to work with the newest version?
Thanks.

raoulteeuwen, 2020-04-03:
- what bin exactly did you flash?
- can you check whether mDNS (<ip>/settings/wifi-setup) and NTP (<ip>/settings/time) are switched off, and if not, switch those off? If that doesn't help, also check (and if on switch off) MQTT
- possibly a proper erase and flashing of the bin 'fresh' could help if that is possible
- afaik a new version is on the horizon; you might consider waiting for that?

aryedevops, 2020-04-03:
Thanks for your reply.
I used the same as I did with the older version Plain `WLED_0.9.1_ESP8266.bin`
I will have to check it on the older version 0.86 since with the newer version it is not possible to do - no enough time to check and then need a reboot - Power off and Power on back.
This is what I have now (0.86) 
[image]
And NTP:
[image]
So it seems like there is no NTP and I have only left the default for the mDNS (of which I don't have and it is not working for me http://wled.local NO Good)
What should I do? Thanks

aryedevops, 2020-04-03:
Just did a fresh install on another ESP8266 and I have the same issue - getting RED Error lines with connection problems. I wanted to use it and also have the MQTT as well. I am Now getting the issue with no extra settings but the default one - no MQTT settings at all and still losing connection every 10 seconds. What can be the problem? thanks

mike2nl, 2020-04-03:
I did a fresh install from the DEV version yesterday. NO issues at all.
- no reconnections
- mqtt working

But i use a fixed IP address. Check your router log for issues.
Have you done the new firmware via serial flash?
Which old version was installed?
Have you done a clear flash once before uploading via serial?
Have you used OTA from an older version?

I use the Oririginal Wemos/Lolin D1 Mini from the Lolin ali express store. 
I never used a clone in any way. 
- check the voltage regulator for the on top printed name
- check if he gets warm

aryedevops, 2020-04-03:
I did flash erase and right after it uploaded the firmware - that is how I work all the time.
```
esptool.py.exe --port %1 erase_flash
esptool.py.exe --port %1 --baud 460800 write_flash --flash_size=detect 0 A:\Arduino\ESPEasy-New2020\bin\WLED_0.9.1_ESP8266.bin
```
It works Right after it. And I could access the WLED-AP and set my LAN AP credentials and it connected also with a static IP that I set for it.
That is the basic flashing that I do all the time and I have done that till version 0.8.6 - Everything works! Now with the major firmware update I did the same with no luck. 
It works and Freezes every now and then - then it gets disconnected every couple of seconds. forcing me to power it off all the time. It seems like the major issue is with syncing and that is NO good for me.
Mine are also WeMOS D1 Mini from Lolin directly.
I have tried two ESP8266 with the same result - went back to version 0.8.6 and it works wonderfully! Not sure why is that? 
I have also tried to upload OTA and it went badly (not sure I knew exactly what firmware to upload since I tried the OTA version first and I lost it right after that - Then I did it Serially again using the code at the top.)
So, It seems like there is an issue with my board and the latest FW that I wish it would be resolved. It is NOT memory issue since these ESP8266 were Tasmotized and worked wonderfully. 

I wish I knew what is the problem? 
Thanks

raoulteeuwen, 2020-04-03:
You say: "It seems like the major issue is with syncing" > can you elaborate?

Could trying the WLED_0.x.x_ESP8266_1M_ota.bin fw-version be an idea, just to see whether that has the same problem?

mike2nl, 2020-04-03:
@aryedevops 
in the case that the tip from @raoulteeuwen is not working 
i would compile for you the latest DEV version of WLED.

If you will go that way please contact me in the discord channel
because i will not share any public stuff here.

Me on discord: mike2nl

austwhite, 2020-04-03:
If you are using a D1 Mini it has 4mbyte flash so you would only use either WLED_0.x.x_ESP32.bin or if it is the ESP8266 version you would use WLED_0.x.x_ESP8266.bin.

The smaller versions such as the ota versions are designed for the ESP-01 boards.

aryedevops, 2020-04-04:
@raoulteeuwen
> You say: "It seems like the major issue is with syncing" > can you elaborate?
> 
> Could trying the WLED_0.x.x_ESP8266_1M_ota.bin fw-version be an idea, just to see whether that has the same problem?

I have already tried the OTA version with no luck and also the 1M as well. I thought that I was wrong doing that after I checked the info about them and realized that there are for a different board - even though they are basically the same chip when deal up to 1M (am I right?)

When I say "major issue is with syncing" is that I get Red lines for losing communication while working on the device from the Web Interface. It looses communication and I am not even knowing that only after I try to save changes.

@mike2nl 
> @aryedevops
> in the case that the tip from @raoulteeuwen is not working
> i would compile for you the latest DEV version of WLED.
> 
> If you will go that way please contact me in the discord channel
> because i will not share any public stuff here.
> 
> Me on discord: mike2nl

I will probably go with your suggestion - Thank you Mike - will message you up in discord.

@austwhite 
I tried the OTA and it did not worked for me at all. I tried it Serially - If I want to be able to use it from the Web Interface is that also a way to upgrade? and If so, How? thanks

Thank you All

aryedevops, 2020-04-04:
@mike2nl 
Sent you a Discord message. Thanks

hediondo1, 2020-04-11:
i suffered the same issue.fixed it changing router wireless channel .

Dreamoffice, 2020-04-13:
same issue here with magichome RGBW Device.

r100gs, 2020-04-13:
I have the same problems now using a esp32. Fresh flash, but nothing helps

Dreamoffice, 2020-04-13:
after a factory reset it is fine as long as i don t connect to my wifi. as soon as i connect the boot loop starts

alexdshop, 2020-04-15:
I have same situation as a @Dreamoffice with one of my four WLED 0.9.1 devices.

triggerx, 2020-04-21:
Also having this issue of the WLEDs (2 of them, ESP8266) coming and going on the network. Getting red bar connection errors.  All started after OTA upgrading them from 0.8.7 to 0.9.1.  Came here to join in on updates to this issue.  thx.

closed 2020-08-27 by stale[bot]