wled/WLED #823: [IDEA] - Apple Home Kit Integration

stockklauser, 2020-04-04:
Hay Guys,

we can try to integrat the HomeKit Integration to Alexa, then it is possible to control it also from the Apple World. 
Im not shure, but i think its only possible on the ESP32 (RAM and Flash) 

https://github.com/maximkulkin/esp-homekit

skbeez, 2020-04-08:
I think this is a great idea however we already have a way for homekit to interact with homebridge.  
I think all we need is an interface between homebridge and WLED. It could just be a homebridge plugin that's needed as the homebridge can mimic whatever command is needed.

gilshahar7, 2020-05-08:
> I think this is a great idea however we already have a way for homekit to interact with homebridge.
> I think all we need is an interface between homebridge and WLED. It could just be a homebridge plugin that's needed as the homebridge can mimic whatever command is needed.

I have a homebridge setup but i i also have a WLED strip at my parents house and i wish there was a way to add it without a homebridge server.

This is certainly possible, even if it would be just an on/off + brightness switch.

ceeeekay, 2020-07-30:
I've posted this elsewhere, but https://www.npmjs.com/package/homebridge-wled-simple does the trick. Basic controls plus colour picker.

closed 2020-12-05 by stale[bot]