wled/WLED #824: APA102 RGBW with separated LEDs support

lwalpurgis, 2020-04-05:
Hello,
At first a very big thank to @Aircoookie , for this really amazing project.
I have a special APA102 RGBWW LED stripe, where an APA102 RGB and a solid WW APA102 led is alternately on the strip. See the picture.
[image]
I used my own Fastled code for this project where I created 2 "virtual" led stripes (RGB and ww) and mapped these at the end before show method in the right order into the real LED stripe, which works good. (I copied the warm white value into all 3 bytes of the warm white led) But now I want to use wled, which is way better. How can I achieve this? Thanks for help!!!

thedannymullen, 2020-04-10:
@lwalpurgis  did you custom build that ?  Or was it an off the shelf item?

lwalpurgis, 2020-04-10:
> @lwalpurgis did you custom build that ? Or was it an off the shelf item?

I ordered this led stripe in this configuration in china...

thedannymullen, 2020-04-10:
Really mice!  I will look for it if you have a link it would be great.

closed 2020-08-15 by stale[bot]