wled/WLED #826: Color fade only works on first segment

jwingefeld, 2020-04-06:
I just realized that the color fade which transitions from one color to another only works on the first segment. The second segment switches to the new color instantaneously while the first segment slowly fades over.

WLED Version: latest version from master branch.

I'll provide a video in discord.

Aircoookie, 2020-04-06:
Thank you!

Yes, unfortunately the transition logic is old and doesn't support any segments but the main segment (the first segment `0` by default).
I plan on replacing it with "transitional effects" in the future, which will also support smoothly fading between different effects!

DarkFox, 2020-08-05:
Have you gotten any closer to implementing transition effects? Otherwise, it might be worthwhile to fix the regular transition for segments.

timhere, 2021-03-10:
I've just encountered this issue, I'm trying to connect to WLED using the json api, and no matter how I try the transitions are only applied to the first segment. For example:

```json
{
	
	"tt": 10,
	"bri": 255,
	"seg":[
		{"col":[[255,0,0]]},
		{"col":[[0,255,0]]},
		{"col":[[0,0,255]]}
	]
}
```

I see a gentle transition to green on the first segment, but the following two just switch immediately. 

As a different approach I also addressing individual LEDs on a single segment:

```
{
	
	"tt": 10,
	"bri": 255,
	"seg": {
		"i": [
			0,10,[255,0,0],
			10,20,[0,255,0],
			20,30,[0,0,255]
		]
	}
}
```

...but still no luck! In this case all LEDs switch immediately.

Another workaround I've considered is creating a preset for each state I need, but I'm not sure that's an optimal solution.

@Aircoookie it's great that this is on your horizon! How do you plan for transitional effects to work when interacting over the API?

Aircoookie, 2021-04-06:
An update on this: In the new v0.12.0 release, color and brightness transitions are now supported on all segments! Effect transitions are still not supported though, therefore this issue will stay open.

timhere, 2021-04-06:
Great news!

blazoncek, 2021-10-03:
This is now supposedly fully fixed in 0.13-b3 including segment transitions. Closing.
Feel free to reopen if you find transitions not working correctly still.

Aircoookie, 2021-10-03:
Thank you, let's keep this one open though. Brightness, color and on/off are now fully supported, but not yet smooth transitions between different effect modes.

blazoncek, 2023-08-05:
I managed to get effect transitions working. 😀

closed 2023-09-05 by softhack007