wled/WLED #829: usermod add func turns light

mehranviva, 2020-04-09:
Hi, thank you for your good project
Tips for adding a feature that I can do by reading the status of the pins
I'm new in programming and I wanted (for etc) running effect and color to be executed when the pin was low, and go back to the last effect, speed, settings,... that were running when it was high.
I can apply the changes, but I could not return to the previous state
sry for my bad english thx for ur help

Aircoookie, 2020-04-10:
Hi!
What is the usermod code you have right now?
To save the old config, try `savePreset(8, false)` and then `applyPreset(8)` to restore. This does work in current source, but not yet in 0.8.5. If you have a reason for still using 0.8.5, you can make it compatible by replacing in `wled01_eeprom.ino`:
```cpp
void savePreset(byte index)
```
to
```cpp
void savePreset(byte index, bool persist)
```

and then at the end of the `savePreset` function add an `if (persist)` in front of `commit()`.

I hope you can get it to work :) Have fun with WLED!

mehranviva, 2020-04-10:
Thanks for your reply
I enclose the code I wrote
The settings return to the previous state but dont updated on the device and the previous settings are still running

[usermod.txt](https://github.com/Aircoookie/WLED/files/4462704/usermod.txt)

Aircoookie, 2020-04-10:
Here, I modified your code a bit. Maybe it works better now:

```cpp
int userval0;

byte ucolS[]{100, 100, 150, 0};                  

byte ueffect;
byte ueffectSpeed;
byte ueffectIntensity;
byte ueffectPalette;
bool trig = true;

#define RIGHT_IN 13

void userSetup()
{
  pinMode(RIGHT_IN,INPUT_PULLUP);
}

void userConnected()
{

}

void userLoop()
{
  userval0 = digitalRead(RIGHT_IN);
  if(!userval0)
  {
    if (trig)
    {   
      for (byte i=0; i<4; i++)
      {
        ucolS[i] = col[i];
      }
      ueffect = effectCurrent;
      ueffectSpeed = effectSpeed;
      ueffectIntensity = effectIntensity;
      ueffectPalette = effectPalette;
      
      effectCurrent=40;
      effectSpeed=240;
      effectIntensity=230;
      effectPalette=0;
      
      colorUpdated(5);
      
      trig = false; 
    }

  } else {
    if (!trig)
    {
      for (byte i=0; i<4; i++)
      {
        col[i]=ucolS[i];
      }
      effectCurrent=ueffect; 
      effectSpeed=ueffectSpeed;   
      effectIntensity=ueffectIntensity;
      effectPalette=ueffectPalette;
      
      colorUpdated(5);
      
      trig = true;
    }
  }       
}
```

mehranviva, 2020-04-11:
Hi thx a lot its working good and my next step is going to add some setting in web ui.
How can I add a button like “2nd color” that when turned on, affect the change only on the usermod variable
Thats what i mean
      `effectCurrent=40;
      effectSpeed=240;
      effectIntensity=230;
      effectPalette=0;`
Changing the 40 240 230 0

Aircoookie, 2020-04-11:
Hi, unfortunately making changes to the web ui is not that easy: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality#web-ui
And you'd also need to extend the HTML API in `wled03_set.ino` for your custom values.

A way easier method is to just save your
`effectCurrent=40; effectSpeed=240; effectIntensity=230; effectPalette=0;`
in a preset (the 1-16 save slots). Then you can always set new values in the UI by overwriting that preset. Your usermod then just applies the preset if the pin goes low and restores them if it is high agai.

mehranviva, 2020-04-11:
Thats good idea to use presets
And how i can set preset in usermod? Is this working `applypreset(n)` in 0.8.6?

Aircoookie, 2020-04-11:
Yes, `applyPreset(n)` will work :)

closed 2020-04-15 by mehranviva