wled/WLED #831: 2 separate sets having lag issues and can't connect

ikrowni, 2020-04-10:
I have 2 separate runs of led strips each with their own esp8266. I'm on the latest build on both. They have been running great until today. I noticed that the lights are kind of lagging. Like I for example I have the drip effect going down my office desk legs and about halfway down the animation just stops for a second then continues. Then I noticed my stair lights weren't turning off so I tried logging on and it can't find the device. Tried unplugging both devices and after plugging them back in they're still having the same issue. This just started today and it's happening with both of them after running just fine since the latest update. 

Looking at the support section it says the lag can be caused by too many led lights but I only have about 200 per run.

raoulteeuwen, 2020-04-10:
Could it be a connection issue, since 'everything is the same as before, it did work fine, suddenly it has problems'. Like you bought a new wireless device that is interfering, you live near other people that might have bought a new access point that interferes etc? Any way to move the devices closer to your AP or have a mobile device act as hotspot so your devices use that as AP and see with another device whether you can than access them?

Did you also check on your router whether your devices show up with an IP? They devices are not in 'AP-mode' (no new AP you can see in your list of AP's)?

Any other specifics for your config that might be overloading the devices, like using MQTT or other interfaces that for some reason have gone haywire and are causing delays etc?

ikrowni, 2020-04-10:
Well I left them powered off completely overnight and after plugging them back in today they seem to be working again. I don't think it was a connection issue my office desk Lights have the esp8266 about 2 feet away from the router. I have 6 Google WiFi pucks in a 2000sq ft home. No new devices have been added lately. I wouldn't have found it so strange if it had just been one of the devices but both having the same like hiccup and no ability to access the UI was weird.

closed 2020-06-01 by ikrowni