wled/WLED #832: compile fails with #define WLED_ENABLE_DMX

pille, 2020-04-10:
when `#define WLED_ENABLE_DMX` is used, the build fails since the cpp-refactoring
```
wled00/wled.cpp: In member function 'void WLED::setup()':
wled00/wled.cpp:216:3: error: 'dmx' was not declared in this scope
   dmx.init(512);        // initialize with bus length
   ^
*** [.pio/build/d1_mini/src/wled.cpp.o] Error 1
```

Aircoookie, 2020-04-10:
Fixed that just today! Try if latest master works.

> maybe we should ensure that travis builds all features.

yep, i should add one environment that has all features enabled, not just the default ones

pille, 2020-04-10:
you're right, i was on an older branch.

OT: what's your plan regarding #817? i'd like to work on DMX output and this blocks it currently.

closed 2020-04-10 by pille

Aircoookie, 2020-04-10:
I'm still waiting on feedback on 817. Will probably merge it though soon!

jwingefeld, 2020-04-11:
just punched in my feedback for 817. :)

pille, 2020-04-20:
PR #858 extends travis tests