wled/WLED #838: Request: 28BYJ-48 motor with ULN2003 support

ozayturay, 2020-04-11:
I have made this project and want to use WLED with it but it lacks the motor control feature for rotation. Can somebody please implement it. Thank you. :)

https://www.youtube.com/watch?v=duiAbLsXZSE

chiefymuc, 2020-04-13:
Hi,
I had the same idea and just put this together.

copy-paste into usermod.cpp.

I'll do a PR soon :)

`#include "wled.h"
#include <CheapStepper.h>

CheapStepper stepper (14,12,13,15); 
// GPIO14 = D5 <--> IN1
// GPIO12 = D6 <--> IN2
// GPIO13 = D7 <--> IN3
// GPIO15 = D8 <--> IN4

unsigned long moveStartTime = 0; // this will save the time (millis()) when we started each new move
boolean moveClockwise = true;

/*
 * This file allows you to add own functionality to WLED more easily
 * See: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality
 * EEPROM bytes 2750+ are reserved for your custom use case. (if you extend #define EEPSIZE in const.h)
 * bytes 2400+ are currently ununsed, but might be used for future wled features
 */

//Use userVar1 (API calls &U1=, uint16_t) to set relay timer duration

//gets called once at boot. Do all initialization that doesn't depend on network here
void userSetup()
{
  stepper.setRpm(20); // Initialize at 20 RPM (might be a lot slower due to WLED running!)
  userVar0 = 1; // Mode - 0 = Single move counterclockwise / 1 = Single move clockwise / 2 = Keep running counterclockwise / 3 = Keep running clockwise / 4 = Back-and-forth forever / 5 = Set RPM (between 10 to 22 recommended) / 6 = stop stepper
  userVar1 = 0; // Degrees (or set Speed when userVar0 is at 5)
}

//gets called every time WiFi is (re-)connected. Initialize own network interfaces here
void userConnected()
{

}

//loop. You can use "if (WLED_CONNECTED)" to check for successful connection
void userLoop()
{
  // we need to call run() during loop() 
  // in order to keep the stepper moving
  // if we are using non-blocking moves
  
  if (userVar0 != 6){
    stepper.run();
  }

  ////////////////////////////////
  // now the stepper is moving, //
  // let's do some other stuff! //
  ////////////////////////////////

  // let's check how many steps are left in the current move:
  
  int stepsLeft = stepper.getStepsLeft();

  // if the current move is done...
  
  if (stepsLeft == 0){
    
    // Mode 0 = Single Move Counterclockwise
    if (userVar0 == 0){
        moveClockwise = false;
    }
    // Mode 1 = Single Move Clockwise
    else if (userVar0 == 1){
        moveClockwise = true;
    }
    // Mode 2 - Run forever Counterclockwise
    else if (userVar0 == 2){
        moveClockwise = false;
        userVar1 = 10;
    }
    // Mode 3 - Run forever Clockwise
    else if (userVar0 == 3){
        moveClockwise = true;
        userVar1 = 10;
    }
    // Mode 4 - Back and forth
    else if (userVar0 == 4){
        moveClockwise = !moveClockwise; // reverse direction
    }

    stepper.newMoveDegrees (moveClockwise, userVar1); // move X degrees from current position
    moveStartTime = millis(); // reset move start time
    
    // Turn of new move if in mode 0 or 1
    if (userVar0 == 0 || userVar0 == 1){
        userVar1 = 0;
    }

  }

  // Mode 5 - Set RPM (change userVar to 6, then userVar1 to RPM (10 to 22), then userVar0 to 5)
  if (userVar0 == 5){
    stepper.setRpm(userVar1);
    userVar0 = 6;
    userVar1 = 0;
  }

  // Mode 6 - turn off stepper
  if (userVar0 == 6){
    stepper.stop();
    userVar1 = 0;
  }

}
`

Have fun! Additions / Suggestions welcome!

ozayturay, 2020-08-14:
@chiefymuc Thank you. I'll try ASAP. :)

I'm closing the issue for now.

closed 2020-08-14 by ozayturay