wled/WLED #839: Feature request: numeric value for speed,intensity,luminosity,rgb, white point

Maurotb, 2020-04-12:
Hi, i have two strip led, i need to make same colour, speed ,intensity, luminosity for this strip.
Is difficult to make identical value with colour palette and slide bars...
Another feature that serves, in addition to the color palette, it would be useful to have also a palette for white (with various shades from cold white to warm white) such as philips hue

Aircoookie, 2020-04-13:
Hi! You can automatically sync both strips together using the sync button in the top of the UI.
But I agree that a numerical input could be helpful!
Color temperature/white shade input is also planned :)

Maurotb, 2020-04-13:
I need to power on/off two strip independently, sync is not an option for me :)
Good! I'm waiting new release, great wled !

closed 2020-04-13 by Maurotb

Aircoookie, 2020-04-13:
You can just turn the sync button on, set the correct palette/sliders and then turn it off again.
Then both will be controllable independently again :)