wled/WLED #840: Stepiness brightness fading, is it possible to smooth it?

StefanRu1, 2020-04-13:
Hi,

i have a question, to find out if this is normal behavior or not.
When i switch on the LED stripe from brightness 0 to 255 with a transition time of 30 seconds or longer the brightness fading seems to be done in steps. 
So it is not a smooth fading, there is a bit of flickering or stepiness going on.
The same is true when going from 255 to 0 over 30 seconds or longer.

The effect always occur when the fading time is long enough, in my case 30 seconds.
I tried to film it, but it is hard since the camera is smoothening the flickering out.

This is not a real issue, but kind of a behaviour which could look better.

Is this the expected behaviour? Is there something i can do against it?

Thanks a lot,
Stefan

My setup:
3 Lights with Wled 0.9.1
1x 146 LEDS WS2812B IC 60LED/M and 8 A   (Floorlight)
1x   77 LEDS WS2812B IC 60LED/M and 8 A   (Kitchenlight)
1x   96 LEDS WS2812B IC 60LED/M and 3 A   (Stimmungslicht)

All 3 show the same symptom.

Aircoookie, 2020-04-13:
Hi Stefan,
unfortunately this behavior is standard on most addressable LEDs (and probably all addressed by WLED). There are only 256 possible brightness values per channel, and with slow transitions (> 5 secs) this becomes very noticeable, especially on lower brightness values.
There is a technology called dithering which could make the "steppiness" better but costs performance and could cause more flicker (it basically rapidly switches between two adjacent brightness values), so I'm not planning on adding it for now.

The only thing you could really do about it is use a quicker transition, at least for about brightness 0-64...

StefanRu1, 2020-04-13:
Ok, perfect. 
I just wanted to rule out that it is a electric problem and i could smooth it somehow.
It is not a problem and just a minor cosmetic issue.
Thanks for the explanation.

Have a nice easter holiday and thanks again for the great software.

Best regards,
Stefan

Aircoookie, 2020-04-13:
Have a nice Easter holiday too :)

closed 2020-04-15 by StefanRu1

phoenixtechnam, 2023-05-02:
I have had that issue for a long time on both v0.13 and 0.14. I thought it was an issue with resources on my ESP8266, but my new ESP32 from QuinLED actually does the same thing. As can be seen in the video, the the leading LEDs are not faded but the trailing LEDs have nice and smooth fade-out. This behaviour can be seen in almost all moving effects at very low speeds.
Effect shown: Sinelon Dual

https://user-images.githubusercontent.com/29205326/235634219-dd650900-808d-4c7f-a8f1-8448e8ef4112.mov

Kindly assist. I tried all versions including nightlies (0.14-b2) and also MoonModules v0.14.0.b15