wled/WLED #841: Status LED outside?

Beliar-666, 2020-04-13:
Hello,
not really an issue, but i wanna know if it is possible to wire an single LED for Status on one of the GPIOs? Ste Nodemcu is in a housing, so i can not see if its powered or not, so i like to connect a seperate LED to one of the GPIOs to get the status Control outside the Housing? Is this possible?

Aircoookie, 2020-04-13:
Hi, yes, this is possible! If you just want to have power state, just connect an LED and appropiate resistor between 3.3v and gnd pins. You could also connect an LED between 3.3v and gpio6, then it will turn off if the lights are on. (alternatively, connecting between gpio6 and gnd will turn the LED on when the lights are on)

Beliar-666, 2020-04-15:
Nodemcu V3 has no GPIO6. https://i1.wp.com/www.teachmemicro.com/wp-content/uploads/2018/04/NodeMCUv3.0-pinout.jpg?resize=768%2C507&ssl=1 So which pin do u mean?

Aircoookie, 2020-04-15:
Oh, I'm sorry. I meant D6 (gpio12). I want to have a serious talk with the people who decided to number the pins differently...

Beliar-666, 2020-04-15:
Yes, thats a crazy thing. Now it works.

Beliar-666, 2020-04-15:
[https://www.facebook.com/photo.php?fbid=10217213813043870&set=a.1214378366075&type=3](url)  This is the Project i´m actually working on.... ;-)

codmpm, 2020-07-03:
Is it possible to re-assign the status led pin? 
I have a custom board with an esp-wroom-02 which does not bring an own led. Mine is on IO12...

Aircoookie, 2020-07-05:
@codmpm there is no proper status LED support in WLED for now, that's something on the list to add!
The current "status LED" just reports if there is power - you could just connect the LED between ground and 3.3v.
However, gpio12 by default is configured as an output originally meant for a relay, but if an LED is connected to it it will be on if the main LED are on!

codmpm, 2020-07-06:
Thank you! Yes, the LED is on "all the time", so it will stay this for the time being.

closed 2020-11-11 by stale[bot]