wled/WLED #842: LED Count mismatch

Beliar-666, 2020-04-13:
Hi, now a real issue i can´t figure out why this happens. I have a strip with 150 LED Its a 2813 (The one with the reserve Signal line). It works fine but one Problem exists.
WLED version 0.9.1

I have to set number of Leds to 151 if i set it to 150, what i really have, the Last Led stays off and wont light up. Any Idea why this happens? Why this is a Problem? If i use e1.31 to control the Leds, the pattern doesnt dit, cause the Count is wrong. Hope u understand what i will say.

Any Idea why this happens? Why i have to Config one more Led than i have?

Aircoookie, 2020-04-13:
Hmm I have no idea why this happens... you could try a slow Scan effect to see which LED is skipped/counted twice. As a workaround, try setting 150 in config and checking "Skip first LED", that should fix the E1.31 issue.
Might be related to the backup pin or something...

tonyn0, 2020-04-13:
Maybe it's just like the wonky numbers in segments?

closed 2020-08-19 by stale[bot]