wled/WLED #843: Connectivity and Stability Issues HA 108.3 and WLED 0.9.1n

SawKyrom, 2020-04-13:
I am having serious connection stability issues with WLED on a NodeMCU 1.0 Amica ESP8266
The concept and application is really nice, however I've spent days and cannot get WLED to remain connected and functioning.  Has anyone else experienced this issue or am I alone?  

System:  HA 108.1, python 3.7.7, WLED integration install
WLED: 0.9.1n installed via Arduino .ino file
Router: ASUS RT-86U Merlin (firmware current update)

Here is the router log file showing frequent disconnection:
[image]

Also, the WLED NodeMCU does not register an IP address, even when it is connected and I can access it through the webpage using the static IP assigned during initial AP setup:
[image]

Here is the Arduino serial output logs:
[WLED Arduino Serial File.txt](https://github.com/Aircoookie/WLED/files/4471766/WLED.Arduino.Serial.File.txt)

I have reinstalled/flashed the NodeMCU several times with the erase flash (sketch and wifi) setting, but I can not get a stable version.  Also, changing the wifi settings in the wled.h file prior to flash does it not improve the connection or actually update WLED settings.  Example: I've changed the static IP in the wled.h file and still only accessible from the old web server IP address:
[image]

Any help/suggestions would be greatly appreciated.

SawKyrom, 2020-04-13:
The issue appears to be related to the HA setup.  I disabled the integration, deleted some WLED MQTT entities, and monitored the Arduino serial output.  The low heap error and data dump disappeared from the log files. I then rebooted HA, reinstalled the integration, and kept the system simple - No WiFi signal strength entity or MQTT status request.  So far, there is a major improvement in stability.  I will monitor for a day or two and report back of any changes.  I suspect that there was a conflict with MQTT calls and/or some entity settings that I had previously set.  

My router still does not report an IP address for the WLED NodeMCU when connected???  This appears to be new with the most recent WLED firmware.  Can you verify?

Aircoookie, 2020-04-13:
Hi,
sorry for the issues!
Are you using MQTT? The native HA integration does not require it, maybe turning it off in WLED helps. Also, what is your LED count?
I cannot imagine it being a major issue, but does changing Flash Size to 4M (1M spiffs) change anything?
Just checked, my router does correctly display the statically set IP in the latest source.
Hope we can get it to work :)

SawKyrom, 2020-04-13:
Aircookie, thank you for the awesome program!  I think the issues were a result of me getting too complex with pulling WLED status data (i.e. signal strength, current status "online/offline", etc.) for monitoring purposes.  Many of these entities are disabled in the integration setup.  Is there a stability issue with those and reason to avoid?

The setup is 894 LEDs powered with two 50A power sources.  I have too much voltage drop from wire resistance to cover the distance even with power injection lines every 5m.  Two power sources on the terminal ends, sharing only common ground.  The NodeMCU has steady 5.02V VIN, so I ruled out dirty power.  The NodeMCU is close range to router and when I did have your signal strength entity enabled, was getting 60-70%.  

Regarding disabling MQTT, I notifed that operation within the integration sends a MQTT calls.  Would there be conflict with HA generated MQTT commands?  A post from last year suggested that the MQTT settings were getting an update and many of the previous ones from 0.8.6 were disabled.  

I had the Flash Size set to 4M.  No change.  What do you make of the "Not-Found HTTP call:" error? Also, still unsettled why my router is showing IP address "offline" (see pic) with that MAC even though it is connected and responsive.   

Thanks again for the quick reply.  The features of your program really are great! Hope to get it stable.

cbourne142857, 2020-04-13:
I have a similar issue - I suspect it is to do with something in the latest Merlin firmware.

Brief notes:

I'm running a D1 Mini - has been working well for about 5 months prior.
No MQTT (anymore) using the homeassistant integration
Initially suspected a power issue - but I have the same problem with lights disconnected/different power source
It is stable in AP mode, but bootloops when connected to my Asus router (AC68U, Merlin 384.16)
If I refresh enough, I can get the web interface to load between reboots
I *think* the issue started when I updated the router from 384.14 to 384.16
I'm flashing with ESP Easy Flasher

I've tried a few other versions - and have the same issue (0.9.1/ 0.9.1 1M / 0.8.6 / 0.8.5)
I've flashed a blank 4M binary to it to see if that helps - it didn't
The same hardware is stable with Tasmota

My wife is working from home today, so I won't be popular if I downgrade the router firmware right now to try that.

SawKyrom, 2020-04-13:
Well, I just watched it disconnect on the log files after 30 minutes uptime, so it appears that there is still a connectivity issue with my setup and has to do with the low Heap size.  It did correspond with me making an effect selection within the integration.

System log:
17:50:22.973 -> Loops/sec: 1666
17:50:27.977 -> **Heap too low! 8144**
17:50:27.977 -> Forcing reconnect.
17:50:27.977 -> state: 5 -> 0 (0)
17:50:27.977 -> rm 0
17:50:27.977 -> pm close 7
17:50:27.977 -> Access point disabled.
17:50:27.977 -> Connecting to 86...
17:50:27.977 -> sleep disable
17:50:31.391 -> scandone
17:50:32.276 -> state: 0 -> 2 (b0)
17:50:32.276 -> state: 2 -> 3 (0)
17:50:32.309 -> state: 3 -> 5 (10)
17:50:32.309 -> add 0
17:50:32.309 -> aid 13
17:50:32.309 -> cnt 

@cbourne142857  Thanks for the information.  Sounds like we have similar problem.  I have several other NodeMCUs running security reed switches and I don't think I've had any connectivity issues with my Merlin firmware and these devices.  But, I'll have to check to be certain.  I've been at this for days and I think I'm going to step back for a bit a reassess.  Also, thanks for mentioning the other versions and blank 4M binary flash.  That was my next move, but no need now.  

Good call on not flashing your router firmware right now.  Domestic violence is at an all time high these days, and I'm sure that wouldn't help your cause.

cbourne142857, 2020-04-13:
Thanks for looking out for my wellbeing ;)

I think I'm getting somewhere - I turned on mobile hotspot on my phone and let the WLED device connect to that instead and it is stable.

marcoboutch, 2020-04-14:
Hi, i have wifi connectivity issue before really long time with WLED. But for an other project, i have some instable problem with MQTT and library <PubSubClient.h>. I dont know if this library is use in WLED but sometime, MQTT refuse connection and loop on the mqtt connection and never connect. Just a clue for WLED dev if no problem whitout MQTT. i can see someting. NodeMCU work ok with WLED. I powerdown my AP. Restart AP. Impossible to ping this NodeMCU. But i see connected in my AP by MAC address.... In my case i will try use static ip because now i use reserved ip in my DHCP. Sorry for my bad english. ;D

r100gs, 2020-04-14:
I have the same problem, but dont use MQTT.
MQTT is dissabled in all my WLEDs but I dont get a stable connection atm.
HA is at 0.108.3

raoulteeuwen, 2020-04-14:
Fyi, i've opened an issue with the Home Assistant team. Since it worked fine till 0.107.7 of HA, something in 0.108.x seems to have changed causing this? See https://github.com/home-assistant/core/issues/33921 .

clsferguson, 2020-04-14:
I am having some issues with one wled setup running 0.9.1n.
two other wled setups are running fine. I flashed OTA with arduino. 
I updated from ver 2002252.

but the one setup was having some issues since updating home assistant to 0.108. (which is why i figured i should update wled.)

So the weird part, I did a full wipe and flash on the non working setup, then I changed out the nodemcu entirely. Meanwhile the other two setups are still fine.

right now i am testing the non working setup in ap mode. it seems to be fine right now. I also have the old nodemcu powered by usb connected and running 0.9.1n, although I haven't paired it with HA yet.

Kosh42, 2020-04-14:
I'm getting the exact same issue since HA upgrade, however I was getting this intermittently before when messing with effect settings. My issues is replicated exactly in all four WLED instances, as was the occasional glitch before.

Doing a little digging, ESP8266 chips seems to enter a reset cycle linked to the watchdog timer. This may or may not be a blind alley as I know very little.

Re flashing has proved unhelpful, but other firmware works fine. Just WLED that causes the resets.

Kosh42, 2020-04-14:
Update - deleting the integration in HA fixed the issue. Seems Frenk has some work to do instead of Aitcookie.

SawKyrom, 2020-04-14:
It does appear to be a HA and WLED interface issue.  Hopefully @raoulteeuwen HA post will generate some traction on fixing the stability issue if it is HA driven.  I looked at my logs from the last several days, and with the bare bone install/new flash and no activity, they are riddled with api connection errors (see attached for short snapshot). 

[image]

I don't think it is MQTT driven, since I've not performed any actions for the last several days.  I wonder if there is a api conflict is due on old values from previous WLED board flashes or the new HA version???  Whatever the cause, the logs suggest it is a polling conflict since it occurs frequently without use.

Aircoookie, 2020-04-14:
@SawKyrom I would not necessarily say the problem lies with HA. Probably the NodeMCU is running out of memory... To verify that, you could try drastically lowering the LED count (to 50 maybe) and see if the disconnects in HA are gone.

SawKyrom, 2020-04-15:
@Aircoookie I'm not sure either, but I can tell you there are several people with various setups experiencing the same issue.  https://github.com/home-assistant/core/issues/33921  I have lowered the number of pixels as suggested - no improvement.  Another individual in the HA 33921 post has similar problems with a rig containing only 5 pixels on the board.  Doesn't appear to be pixel number related.  

[image]

cbourne142857, 2020-04-15:
Mine seems to be stable now - I removed all wled integrations from HA, rebooted the server, added integrations again. I'm not 100% why that fixed it, but it seems to have.

austwhite, 2020-04-15:
Try a ticket in the GitHub for the HA integration.
I believe it uses the API which could explain why you had issues.  Random garbage on the API could cause all sorts of weird behaviour.

SawKyrom, 2020-04-15:
@cbourne142857 I concur.  I disabled the HA integration last night and wanted to monitor 12 hours before replying.  I can say with certainty my observations are the same and thus conclude the issue is with the integration.  My setup has been absent of the connection issues ever since.

I am controlling the NodeMCU with strict MQTT commands.  I’ll submit a ticket for the HA integration.  Until then, I would use the dedicated AP connection or MQTT.  Thanks!

r100gs, 2020-04-16:
ATM for me it is working again. Since 1 day everything is running stable now.
1. I deleted all my WLED-Devices from HA.
2. Restart of HA
3. Let HA autodiscover all my WLED-devices again.

Best regards,
Stefan

geertmeersman, 2020-04-17:
> ATM for me it is working again. Since 1 day everything is running stable now.
> 
> 1. I deleted all my WLED-Devices from HA.
> 2. Restart of HA
> 3. Let HA autodiscover all my WLED-devices again.
> 
> Best regards,
> Stefan

Thanks a lot Stefan!! Some of my hairs got grey, until I bumped into your post! This made my WLED work again, cool!

SawKyrom, 2020-04-24:
@r100gs @geertmeersman @cbourne142857 Have any of you had further troubles with your WLED connectivity?  

I am still having issues with WLED connectivity. I had a stable setup for a few days, and then yesterday I activated an older NodeMCU with a pool controller program - the whole thing went crazy again causing the WLED node to drop connection, auto reset, auto on solid colors, etc. This older NodeMCU was a previous WLED ESP8266 with different IP and unchanged MAC. This device was later found to be unstable also.

I noticed on my pool controller that the CWMODE was either AP or both (despite me not setting this function in my sketch)  when I discovered it broadcasting a SSID. I did not setup up the device to be in softAP and thus I presume it to be a carry over from the WLED setup. I had also completely flashed a blank bin, which must have no effect on the ExpressIf ESP8266 wifi settings. @cbourne142857 had mentioned this finding as well.

It is also all but impossible for me to find the NodeMCU 8266 firmware bin from China. I want to completely restore factory settings and remove any other deep level settings initiated by WLED which may be causing the connection problems.  Anyone what to point me in this direction, It would be much appreciated.

I added to my pool controller NodeMCU wifi setup the "WiFi.mode(WIFI_STA);" command to get the device to stop broadcasting an SSID. I'm testing this today for stability. I still don't know if this would be causing the connectivity issues, but I'm trying everything. BTW, my RSSI on this device is mid 70's.  Anyone else know their RSSI values???

gniknalu, 2020-05-05:
I was facing the same situation (none of the WLED's running on my ESP8266's showing up on their http page nor on the WLED app). I deleted the integration in HA and now the WLED app still can't connect but I can via http. Now I'll upgrade to 0.10

Merwenus, 2020-05-10:
Same with Wemos D1 (not mini), remove-reinstall integration fix the problem until reboot or wifi loss (router restart, docker restart etc) and it starts again.

closed 2020-09-14 by stale[bot]