wled/WLED #845: Troubleshooting e1.31 (WLED with Jinx)

DrMarlin, 2020-04-14:
Love WLED and have multiple instances around the interior and exterior of my house. Thank you for the great work.

I'm attempting a project to build an LED Matrix, and I want to use Jinx to control it. Unfortunately I've hit a problem and I'm not sure what is the cause.

The WLED functions fine under manual control via the web GUI. I can change colors, effects, etc.

In Jinx I add the output device for the WLED device -- I've tried both unicast to specific IPs and multicast. 

Like clockwork, as soon as I select "Start Output" in Jinx the lights on the WLED turn off, however nothing I've set up in Jinx seems to go through - no color, nothing.  When I stop outputting, the lights powered by the WLED return, which does suggest it is connecting.

I'm not sure how to debug/diagnose this. 

If there is anyone who has set up E1.31 and used Jinx I'd love to hear your experience and potential advice.

Cheers!

jwingefeld, 2020-04-14:
Did you set the patch matrix in Jinx? This maps all the channels to your actual matrix.

DrMarlin, 2020-04-14:
Aha! So clearly the issue is I'm clueless! Thank you for the tip - its working now.

closed 2020-04-14 by DrMarlin