wled/WLED #846: Would like to see better indication of color source in presets.

JoshWobbles, 2020-04-14:
For example some presets have baked in colors, some use only the 1st custom color, some 2 etc... I would be much easier from a user standpoint to know what pattern will use what colors from where, though the implementation of this may be a bit complex.

Perhaps if a baked in pattern is chosen with no custom colors, the picked color spots on the wheel tab disappear, if the pattern uses 1 color only show 1 pickable color spot, 2 for 2 and so on.

Also would be nice to do the reverse as well, for instance if I choose the colors red/white/blue, it would be nice to have the option to filter the list of presets to those that will use the 3 colors.

I get this involves some re structuring of the data and for me it's easy enough for me to figure out without these enhancements, but if I were to install these on say my fathers patio, he likely wouldn't be able to figure it out.

Aircoookie, 2020-04-14:
Thank you for the suggestion!
This would actually be really useful, I agree. I am planning to add some code to dynamically show and hide sliders to have them only for the effects that actually support them. Colors could then also get an indicator if they are not used by the current effect mode!

pille, 2020-04-14:
this may also be relevant for 3rd-party software, this info should also be provided in an abstract human/machine readable way. the wiki is already a good starting point. this will be useful e.g. in E1.31/ArtNet sources.

JoshWobbles, 2020-04-30:
Seems that changing the data layout to a more serializable form is already mentioned elsewhere, that may be a prerequisite to this as the change from hardcoded structure to a more array-ish type for the colors would be necessarily to avoid hardcoding all these UI changes aswell.

blazoncek, 2022-07-25:
The OP request may be partially resolved by #2610 (hiding unneeded UI elements).

blazoncek, 2023-01-25:
Implemented (to some extent) in 0.14.

closed 2023-01-25 by blazoncek