wled/WLED #847: E131 universe drop-outs on 0.9.1 esp8266

samstreak, 2020-04-15:
I have used WLED for several versions now and not had an issue with my 300-LED strip via xLights. Since I updated to 0.9.1 every few seconds one of the universes drops out - not always the first or second one - it just stops updating while the other one keeps working fine. After another few seconds, the 'missing' universe starts working again, but only for a few more seconds. I thought it was a network issue at first but wasn't seeing any problems there even after removing some of the other network devices temporarily to reduce network chatter. I dropped back to 0.9.0b1 and now the problem is gone. Any ideas on what to troubleshoot next?  The WLED core works fine/steady with the local LED control, it's just when receiving via E1.31 that it does this.

austwhite, 2020-04-15:
I have noticed this issue a little as well.  Thought it was my ESP chip not keeping up so I am waiting for a 32 to come and test as mine are all 8266 ATM.
Watching this to see the outcome

Aircoookie, 2020-04-15:
Hi, thank you for the report! I believe the issue has since been fixed in latest source code. Please try if updating to this binary https://drive.google.com/open?id=1Fm_KPY6pmlRu22wuupHjOx3Z2A4dOmV8
resolves the issue :)

samstreak, 2020-04-18:
Sorry for delay in responding...  looks good so far, thanks for the quick response!

kmccor, 2020-06-12:
Hi, I've been controlling a number of WLED devices via E.131 multicast and I'm experiencing intermittent universe drops outs on version 0.10.0. I can see the server/network side is OK and the multicast data is being generated correctly. I've got a matrix that spans 2 universes and sometimes either half will drop out then after a while start working again, sometimes both universes drop out for a while. A reboot of the WLED device during a drop out always fixes the issue immediately. The same occurs on the WLED devices I've got attached to LED strings.

samstreak, 2020-06-13:
So far on 0.10.0 my unicast (all are unicast at this point) WLED's have been mostly stable... there is some herky-jerkiness but that's just WiFi stability/interference issue in my neighborhood.

kmccor, 2020-06-14:
Quick update - I've flashed Espixelstick to the Wemos D1 Mini hosting my matrix to eliminate network etc. and it's been running over E.131 multicast without dropouts for a couple of days now. Therefore it does appear to be an issue within the WLED software. I'd much rather use WLED given all its capabilities as it's really impressive software so hopefully this can be fixed. If there's anything I can do to provide more info then let me know. Cheers.

Aircoookie, 2020-06-14:
@kmccor Sorry that E1.31 has not been working flawlessly for you in WLED. Since espixelstick works fine, I am pretty sure this is not a hardware or network issue, but rather some other WLED feature interfering. You could try disabling the Alexa emulation and any other interfaces you might had active like MQTT or NTP (in Time settings). How many LEDs are you running? (2 universes should normally be very stable though)

kmccor, 2020-06-14:
No need to apologise! 
I've reverted back to WLED and disabled everything I could aside from E.131 multicast and I'm still getting the dropouts. I hadn't got NTP or MQTT enabled, so I've disabled On/Off button, UDP Realtime & Alexa.

The matrix I'm running is 240 LEDs, although it also happens on the 15, 29 & 35 LED strings I'm also running on WLED.

I never have any problems accessing the web interface during the dropouts, just the Live E.131 splash screen disappears until it reconnects.

samstreak, 2020-10-13:
Right now I don't have time to work on doing a self-compile and I've been
waiting for a new release.

closed 2020-10-20 by stale[bot]