wled/WLED #848: D1 MINI unable to set static IP

Woutch, 2020-04-15:
Dear, I have a D1 mini that I try to flash with the latest WLED software 0.9.1.

This is the regular version of the ESP8266 firmware.

The software boots and I can connect on 1.2.3.4 and set my SSID and password.
The board reboots and gets its IP from my DHCP.

Now I set a static IP and gateway.
IP: 10.0.10.180
NETMASK: 255.255.254.0
Gateway: 10.0.10.1

And the board starts to bootloop:

`[20:56:18]Exception (29):
[20:56:18]epc1=0x4022ab6a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[20:56:18]
[20:56:18]>>>stack>>>
[20:56:18]
[20:56:18]ctx: sys
[20:56:18]sp: 3fffeb20 end: 3fffffb0 offset: 01a0
[20:56:18]3fffecc0:  3ffea1f7 4021a939 00000020 40101510  
[20:56:18]3fffecd0:  40236d14 00000003 3fffed00 00000000  
[20:56:18]3fffece0:  3fffed00 3fff80b4 00000000 4022af53  
[20:56:18]3fffecf0:  3ffea1f7 3fff80b4 00000000 4021a944  
[20:56:18]3fffed00:  006c6100 00000000 00000000 40230d10  
[20:56:18]3fffed10:  3fffed50 3fffed50 3fffbe2c 40230ef4  
[20:56:18]3fffed20:  00000000 3fff80b4 3fffed50 40230b31  
[20:56:18]3fffed30:  00000003 3fff4d08 3fffbe2c 40230b50  
[20:56:18]3fffed40:  ffffffff 00000000 3fffed70 40230b31  
[20:56:18]3fffed50:  3ffed337 3fffed70 00000001 00000002  
[20:56:18]3fffed60:  3ffed337 3fff80b4 4020e6a4 4021aa80  
[20:56:18]3fffed70:  00594e00 00002f6e 8000002f 4022ba59  
[20:56:18]3fffed80:  3fff80c4 3fff80b4 3fff4d54 4022bda8  
[20:56:18]3fffed90:  8025bef4 00000001 3fff80e8 00000002  
[20:56:18]3fffeda0:  3ffed337 3fff4cdc 3fff80b4 4022a151  
[20:56:18]3fffedb0:  0001bd28 00000001 3fff2fb8 00000002  
[20:56:18]3fffedc0:  3ffed337 00000002 3fff80e8 40231bb8  
[20:56:18]3fffedd0:  3ffed337 3fff2f34 3fff5b1c 00000002  
[20:56:18]3fffede0:  3ffed337 00000002 3fff80b4 4022a39c  
[20:56:18]3fffedf0:  00000000 00000000 4bc6a7f0 00000000  
[20:56:18]3fffee00:  4025e235 00000000 40100b84 000010e2  
[20:56:18]3fffee10:  3fff80e8 00000000 3fffb454 4025772c  
[20:56:18]3fffee20:  0013857f 40102fdb 00000020 00000000  
[20:56:18]3fffee30:  3fff7fd4 3fff6cdc 00000000 4022a3f5  
[20:56:18]3fffee40:  3fff7fd4 3fff6cdc 3fff2f20 40227ee8  
[20:56:18]3fffee50:  3fff5f5c 3fff5c90 3fff5c8c 40258892  
[20:56:18]3fffee60:  3fff2fb8 3fff2fac 00000006 3fff2f1c  
[20:56:18]3fffee70:  3fff5f5c 3fff2f24 3fff2f20 40227f2c  
[20:56:18]3fffee80:  3fff5f5c 3fff2f24 3fff2f20 4025c800  
[20:56:18]3fffee90:  00000093 b40a000a 00000018 00000093  
[20:56:18]3fffeea0:  00000000 00000018 3ffef730 40100e04  
[20:56:18]3fffeeb0:  00000000 00000000 3ffec69b 3fff2fb0  
[20:56:18]3fffeec0:  3ffed27e 3fff2fac 3fff6cdc 4025ad65  
[20:56:18]3fffeed0:  3fff2e28 3fff5b1c 3fff5b1c 4010098c  
[20:56:18]3fffeee0:  00000008 3fff5754 3ffef730 3fff5b1c  
[20:56:18]3fffeef0:  3ffed270 00000000 3fff6cdc 4025a29d  
[20:56:18]3fffef00:  5c0a000a 000000f1 00000000 00000018  
[20:56:18]3fffef10:  00000002 0000001a 402507d7 3ffef730  
[20:56:18]3fffef20:  3ffed88c 3fffdcc0 3ffebe10 3ffebe10  
[20:56:18]3fffef30:  4025074a 3ffef730 00000000 3fff5bac  
[20:56:18]3fffef40:  3fffdc80 00000000 3fff6cdc 4025dbdb  
[20:56:18]3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49  
[20:56:18]3fffef60:  40000e19 40001878 00000002 3fffffb0  
[20:56:18]3fffef70:  3fffff10 aa55aa55 000000ed 40105285  
[20:56:18]3fffef80:  4010528b 00000002 3fffffb0 42006574  
[20:56:18]3fffef90:  4010000d 61657220 223d6d6c 69676f4c  
[20:56:18]3fffefa0:  00000000 3fffef3c 00000000 3fffff78  
[20:56:18]3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
[20:56:18]3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
[... 246 lines cut ...]
[20:56:19]3fffff30:  00000000 4bc6a7f0 6b439581 00000000  
[20:56:19]3fffff40:  00000000 00000000 4bc6a7f0 00000000  
[20:56:19]3fffff50:  007a1200 f437304f 40100b84 000010d2  
[20:56:19]3fffff60:  007a1200 f4373fc5 00000000 3fff2798  
[20:56:19]3fffff70:  00000000 3fff24c0 3fff2758 40224805  
[20:56:19]3fffff80:  00000000 00000000 00000001 4010098c  
[20:56:19]3fffff90:  3fffdad0 00000000 3fff2758 401009ad  
[20:56:19]3fffffa0:  feefeffe feefeffe feefeffe 40231f2d  
[20:56:19]<<<stack<<<
[20:56:19]
[20:56:19]last failed alloc call: 4021A939(152)
[20:56:19]
[20:56:19] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[20:56:19]
[20:56:19]load 0x4010f000, len 1384, room 16 
[20:56:19]tail 8
[20:56:19]chksum 0x2d
[20:56:19]csum 0x2d
[20:56:19]vbc204a9b
[20:56:19]~ld`

Anyone have an Idea?

Many Thanks,
Wouter!

Woutch, 2020-04-16:
Flashed with the ESPHome firmware, configured with fastled, no problems.

If I can do any other test, let me know!

Thanks!

Woutch, 2020-04-16:
Also installed platformIO and compiled master, but same error (exception 29),
Altough the stacktrace seems a little different now.

[14:49:17]Exception (29):
[14:49:17]epc1=0x4022839d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[14:49:17]
[14:49:17]>>>stack>>>
[14:49:17]
[14:49:17]ctx: sys
[14:49:17]sp: 3fffec80 end: 3fffffb0 offset: 01a0
[14:49:17]3fffee20:  00000000 3ffe9c45 00000020 3fff26d0  
[14:49:17]3fffee30:  0014d05c 40224f97 0000012c 402330b4  
[14:49:17]3fffee40:  3fff8174 3fff1784 3fffb14c 40229fb9  
[14:49:17]3fffee50:  3fffb14c 3fff1784 3fff569c 40100145  
[14:49:17]3fffee60:  3fff28ac 3fff1784 3fff569c 40224fb0  
[14:49:17]3fffee70:  3fff569c 3fff26d8 3fff26d4 40224fdc  
[14:49:17]3fffee80:  3fff26e8 3fffa674 3fff28b8 402637e6  
[14:49:17]3fffee90:  00000000 b60a000a 00000010 00000000  
[14:49:17]3fffeea0:  00000000 00000010 3ffeee80 40100db8  
[14:49:17]3fffeeb0:  40250000 00000000 00000001 3fff28b0  
[14:49:17]3fffeec0:  3ffed062 3fff28b8 3fffa674 402620fd  
[14:49:17]3fffeed0:  3fff25dc 3fff5474 3fff5474 40100940  
[14:49:17]3fffeee0:  00000008 3fff50ac 3ffeee80 3fff5474  
[14:49:17]3fffeef0:  3ffed054 00000000 3fffa674 4026163d  
[14:49:17]3fffef00:  5c0a000a 00000064 00000000 0000000f  
[14:49:17]3fffef10:  00000002 0000001a 40251e77 3ffeee80  
[14:49:17]3fffef20:  3ffec9e8 3fffdcc0 3ffeb550 3ffeb550  
[14:49:17]3fffef30:  40251dea 3ffeee80 00000000 3fff54dc  
[14:49:17]3fffef40:  3fffdc80 00000000 3fffa674 402577cf  
[14:49:17]3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49  
[14:49:17]3fffef60:  40000e19 40001878 00000002 3fffffb0  
[14:49:17]3fffef70:  3fffff10 aa55aa55 000000ed 40105245  
[14:49:17]3fffef80:  4010524b 00000002 3fffffb0 66666f77  
[14:49:17]3fffef90:  4010000d 6f772f74 2e006666 66666f77  
[14:49:17]3fffefa0:  00000000 3fffef3c 00000000 3fffff68  
[14:49:18]3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
[14:49:18]3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff000:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff010:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff020:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff030:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff040:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff050:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff060:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff070:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff080:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff090:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff100:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff110:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff120:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff130:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff140:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff150:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff160:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff170:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff180:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff190:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff1c0:  feefeffe feefeffe feefeffe feefeffe  
[14:49:18]3ffff1d0:  feefeffe feefeffe feefeffe feefeffe  
[... 209 lines cut ...]
[14:49:19]3ffffef0:  00000000 4bc6a7f0 f0624dd2 00000000  
[14:49:19]3fffff00:  0014bd7c 00000000 4bc6a7f0 00000000  
[14:49:19]3fffff10:  00000001 3fff1cb4 40100b38 00000002  
[14:49:19]3fffff20:  00000000 4bc6a7f0 fdf3b645 00000000  
[14:49:19]3fffff30:  00000000 00000000 4bc6a7f0 00000000  
[14:49:19]3fffff40:  007a1200 cb173474 40100b38 000011ef  
[14:49:19]3fffff50:  00000000 00000000 00000001 3fff1f44  
[14:49:19]3fffff60:  007a1200 cb174590 3fff0d00 4022c651  
[14:49:19]3fffff70:  00000000 00000000 3fff0da8 3fff1f44  
[14:49:19]3fffff80:  00000000 00000000 00000001 40100940  
[14:49:19]3fffff90:  3fffdad0 00000000 3fff1f04 40100961  
[14:49:19]3fffffa0:  feefeffe feefeffe feefeffe 40233365  
[14:49:19]<<<stack<<<
[14:49:19]
[14:49:19]last failed alloc call: 40229FB0(300)
[14:49:19]
[14:49:19] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[14:49:19]
[14:49:19]load 0x4010f000, len 1392, room 16 
[14:49:19]tail 0
[14:49:19]chksum 0xd0
[14:49:19]csum 0xd0
[14:49:19]v3d128e5c
[14:49:19]~ld
[14:49:19]Ada

Woutch, 2020-04-16:
Found a debug version for D1 MINI: Compiled and running AP mode works normally, but after saving wifi settings, back to bootloop:

[15:12:48]---WLED 0.9.1n 2004120 INIT---
[15:12:48]esp8266 2_6_3
[15:12:48]heap 31360
[15:12:48]LEDs inited. heap usage ~144
[15:12:48]Load EEPROM
[15:12:48]Ada
[15:12:48]Access point disabled.
[15:12:48]Connecting to Sas & Wout IOT...
[15:12:51]
[15:12:51]Connected! IP address: 10.0.10.182
[15:12:51]Init STA interfaces
[15:12:52]
[15:12:52]Exception (29):
[15:12:52]epc1=0x4023b08e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

First had mdns set, then bootloop, so tried to disable it, but to no avail, still bootloop but mdns line is gone.

@Aircoookie Let me know if I can do something else?

Best regards,
Wouter

Dareno1995, 2020-04-16:
2units ESP-8266 stopped working out of nowhere. 
Tried whiping flash en installing diffrent versions. 
I removed static IP settings from router and restarted router everything works normal now.

But cant set static ip or bootloop starts.

musicaways, 2020-04-16:
try to leave empty mDNS and then it should work

Woutch, 2020-04-17:
Found the problem, this is a duplicate from #843

I'm using the latest version of HA and when setting the static IP, this ip is configured in HA using the WLED integration, that one apparently has a BUG in the latest version: https://github.com/home-assistant/core/issues/33921

It is hammering the WIFI causing it to rest. 

Confirmed this by deleting all WLED settings from HA and restarting. Now working again.

Thanks!!

closed 2020-04-17 by Woutch

robthepyro, 2020-04-19:
For anyone that had the same problem, I found another solution (and I'm not sure what caused the bootlooping, started as I did WLED 0.9.1 and HA 0.108 at the same time -> mistake) 
I OTA'd one of my wemos D1 minis with 0.9.1 and it bootlooped. Swapped it out with a fresh D1, the same problem! Two other of my existing D1's with WLED upgraded to 0.9.1 just fine however...

I tried erasing flash, then reflashing. Tried 0.9.0b and 0.8.x. Tried deleting all HA WLED integrations. Tried flashing tasmota and espruna, setting those up (which worked) and then erasing + flashing WLED. None of these worked! 

What eventually worked was flashing tasmota, setting up wifi to my ssid, then flashing WLED 0.9.1 _directly_ (without erasing flash!) after which WLED asked for wifi creds and didn't bootloop after I saved them! success! 

@Woutch @Aircoookie , I had both Exception 28 and 29 on both my bootlooping D1's, so I don't know what the problem is with WLED on a blank erased flash, but it was much happier if I flashed over a working tasmota... very interested if this helps identify the bug