wled/WLED #849: WS2801 config

OliPassey, 2020-04-15:
Hey,
Just seen that you've been able to add support for WS2801 thanks to the upgrade of NeoPixelBus
Great news! Im digging out even more LEDs!

Is there a WIKI article that im missing on how to get it going?
I have GND, Data & CLK  connected to GND, 0, 2 (tried GND, 2, 0 also just in case they were mis-labelled) but i just can't get any to light up. Using a 5v 10a supply on a string of 50 pixels. Best result i've had so far is such random flickering when unplugging cables.

It is possible the string has an issue, but they worked last time when used with a raspberry pi / hyperion.

Any steps i am missing, or any options that need enabling in the settings page?
I've used the ws2802 binary from the 9.1 release, and also tried building myself, but the build failed.

Any advice you could offer would be much appreciated.

Aircoookie, 2020-04-15:
Hey,
are you using D0 and D2 or GPIO 0/2. The correct pins are clock on gpio 0 (D3) and data on gpio 2 (D4). Can you verify that?
Hope you can get it to work soon :)

OliPassey, 2020-04-15:
Ah! I definetly had the wrong pins, i will try with those - thanks!

Whilst you're here - im running loads of wled instances and have started 3d printing them into lighting fixtures. Some are battery powered i had been putting my own circuit together but recently found these https://www.ebay.co.uk/itm/Micro-USB-Mobile-Power-Bank-18650-Battery-Shield-V8-for-Arduino-ESP32-ESP8266/274105655348?ssPageName=STRK%3AMEBIDX%3AIT&var=574207736663&_trksid=p2057872.m2749.l2649

Just wondering if you'd had anytime to work on battery status and early shutdown before the cells are drained below a certain point? It would be a huge help to how im using WLED

Thanks as always for an awesome project and the frequent updates!

OliPassey, 2020-04-15:
That fixed it! Thanks!!

Aircoookie, 2020-04-15:
Awesome!

Someone could write a usermod for that, or I could add it to WLED itself - automatic shutoff would be very useful for battery powered projects!

OliPassey, 2020-04-15:
Im not a developer but im able to read and understand the code, would be more than happy to take a stab at it. May not end up code worth of including though :) 
Would be great to see it as a feature either way

closed 2020-04-15 by OliPassey