wled/WLED #852: Not an issue but a rlevant feature request

scargill, 2020-04-16:
WLED is clearly very good at what it does, but one thing stands out. Unlike my own early ESP8266 endeavours and, for example, Tasmota, WLED does not appear to support multiple SSIDs.  After tryng out various modes and having a GREAT time with this in my home office today, I made the mistake of attempting to demo a little board _ LED ring to my wife this evening on her PC. Immediately - a hurdle - my home office is out of range of the house, which is single storey and wide and so has it's own access points. So, my 2 minute demo went no-where - opportunity lost. Simple adding a second access point would further enhance the flexibiliy of this excellent software. I assume that setting up demos in one location and using them in another is commonplace.

chrisl8, 2020-04-16:
I would also love to be able to move my WLED device easily between wifi networks.

I'm willing to dig into the code if someone can point me in the right direction.

scargill, 2020-04-17:
> I would also love to be able to move my WLED device easily between wifi networks.
> 
> I'm willing to dig into the code if someone can point me in the right direction.

I would but I'm a WLED newby - good luck - Happy to be a guinea pig...

Aircoookie, 2020-04-17:
I agree, this could definitely be added, at least for a seond SSID.
The way it would work is to attempt connecting to the primary SSID for the first 10 seconds, then attempting the secondary, then the first again and so forth. The time spent attempting to connect could be lengthened after multiple attempts in case some networks take longer to connect (usually never the case). All the WiFi connection logic is currently in `wled.cpp`.

austwhite, 2020-04-17:
If you do this, we would still need a fall back if it can't connect to both.
Maybe 15 seconds attempt to come next, some APs are slow, and try for 3 times on each SSID then fall back to AP mode?
Just wanted to put my name in to watch this too 🙂

scargill, 2020-04-17:
> I agree, this could definitely be added, at least for a seond SSID.
> The way it would work is to attempt connecting to the primary SSID for the first 10 seconds, then attempting the secondary, then [...]

Sorry about taking so long to come back on this, guys, do I take it we have a winner? 

I'm late as I spent all morning explaining over a Discord video to a pal of mine the benefits of WLED...and sure enough the ONLY minor gripe we mutually came up with was that I could ONLY show him the preview onscreen as I left the actual WLED controller in the house rather than change the access point again just to change it back again for the wife later. 

I note DrZzs did a video on WLED, if this twin access point request goes through, as soon as my 60-LED ring turns up I'll do another video for my own followers - thank you for your response AIRCOOKIE - as someone who's been doing this voluntary stuff for a long time, 

I know how much time even the simplest changes can take, would really appreciate second SSID.. (I did think for Tsamota, but never suggested it to the author, that a rolling list of, say the last 4 used access points would not take too much storage but in this case I'm sure you need as much RAM as you can get for large numbers of LEDs - which brings on a question, I noted this morning the count can go over 256 - but realistically how many LEDS could you handle (ignoring power considerations) - or is that a silly question depending on the effects in use? If that's already documented, just ignore that last question, I'll get there.

AH, I just realised you added it to the 1.0.0 milestone. SUPER.

softhack007, 2023-01-30:
closing as we have newer request with similar topics
* https://github.com/Aircoookie/WLED/issues/2845
* https://github.com/Aircoookie/WLED/issues/2974
* https://github.com/Aircoookie/WLED/issues/2151

closed 2023-01-30 by softhack007