wled/WLED #853: WLED turns itself back on

roblandry, 2020-04-17:
WLED 0.9.1 on ESP8266
Integrated via home assistant 108.4 currently
automation to turn on a preset at sunset and turn off at sunrise via node-red.
Every night at 0300 router reboots Asus RT-AX88U (merlin latest)
I have removed and readded the lights to hass.
Wifi signal:
  * 2cf4327736fc:62%
  * 98f4abb32fc4: 80%
  * 2cf4324b4d9c: 94%.

seems this issue has come up since upgrading hass > 108.
i created a custom component, and added logging to async_turn_on, async_turn_off, and is_on, to see if home assistant is causing the issue.
i have disabled node red to eliminate the possibility of this being the culpret.

The issue:
  * via node red, every morning at sunrise, the lights are automatically turned off.
  the command is getting issued (tried the integration and mqtt T=0 on wled/all/api).
  * via home assistant, manually turning the lights off (via integration).
  * via the ui, master power button.

  * regardless of how i turn them off, they always turn themselves back on.

the only way to turn them off, and keep them off is to reboot the lights, then they work until the next morning. everything works as expected following a reboot.

already rebooted 2cf4324b4d9c, so its working as expected.
2cf4327736fc, 98f4abb32fc4 have not been rebooted. 
This is what the logs show for turning them off via a node red automation, however the method (node-red, integration, mqtt, wled ui) does not matter as the result is always the same.
```
2020-04-17 09:43:12 DEBUG (MainThread) [custom_components.wled.light] WLED turn_off: 2cf4327736fc data: {'on': False, 'segment_id': 0}
2020-04-17 09:43:12 DEBUG (MainThread) [custom_components.wled.light] WLED turn_off: 98f4abb32fc4 data: {'on': False, 'segment_id': 0}
2020-04-17 09:43:12 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4327736fc is False
2020-04-17 09:43:12 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4327736fc is False
2020-04-17 09:43:12 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 98f4abb32fc4 is False
2020-04-17 09:43:12 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 98f4abb32fc4 is False
2020-04-17 09:43:13 DEBUG (MainThread) [custom_components.wled] Finished fetching wled data in 0.217 seconds
2020-04-17 09:43:13 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4324b4d9c is False
2020-04-17 09:43:13 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4324b4d9c is False
2020-04-17 09:43:16 DEBUG (MainThread) [custom_components.wled] Finished fetching wled data in 0.375 seconds
2020-04-17 09:43:16 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4327736fc is True
2020-04-17 09:43:16 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4327736fc is True
2020-04-17 09:43:17 DEBUG (MainThread) [custom_components.wled] Finished fetching wled data in 0.159 seconds
2020-04-17 09:43:17 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 98f4abb32fc4 is False
2020-04-17 09:43:17 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 98f4abb32fc4 is False
2020-04-17 09:43:18 DEBUG (MainThread) [custom_components.wled] Finished fetching wled data in 0.039 seconds
2020-04-17 09:43:18 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4324b4d9c is False
2020-04-17 09:43:18 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4324b4d9c is False
2020-04-17 09:43:21 DEBUG (MainThread) [custom_components.wled] Finished fetching wled data in 0.073 seconds
2020-04-17 09:43:21 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4327736fc is True
2020-04-17 09:43:21 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 2cf4327736fc is True
2020-04-17 09:43:22 DEBUG (MainThread) [custom_components.wled] Finished fetching wled data in 0.049 seconds
2020-04-17 09:43:22 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 98f4abb32fc4 is True
2020-04-17 09:43:22 DEBUG (MainThread) [custom_components.wled.light] WLED is_on: 98f4abb32fc4 is True

```

raoulteeuwen, 2020-04-17:
please see https://github.com/Aircoookie/WLED/issues/843 . seems related? maybe close this issue to spare devs some time?

roblandry, 2020-04-17:
i understand, however #843 is a catch all. i figured i would list the issues i am seeing and lay out everything i have looked at. I didnt see a mention of restarting issues.

raoulteeuwen, 2020-04-17:
o.k., just wasn't sure whether you knew there was a similar issue. Your call ;-).

weini22, 2020-05-03:
I had a (probably) similar issue. When I switchd the device off via MQTT and did a 2nd MQTT switch off after that, the device switched on again.
I solved the issue for me using the HTTP API command within the MQTT interface.
So "wled/cube1/api T=0" is really switching the device off, no way how often you call it.

Aircoookie, 2020-05-05:
@weini22 yep T=2 is a toggle command, it will always change between on and off. T=1 is always on and T=0 is always off :)

closed 2020-09-02 by roblandry