wled/WLED #854: several are flashing off on on to network every 10 seconds or so

stovedoctor, 2020-04-17:
all but one of my WLED on nodemcu are loosing connected to wifi every 10 seconds 
other users on DRzzs chat are seeing the same thing.

stovedoctor, 2020-04-17:
[12:41:54]3ffffcd0:  40251242 3ffef3e8 3fff573c 00000001  
[12:41:54]3ffffce0:  4025134e 3ffef3e8 3fff573c 3ffef3e8  
[12:41:54]3ffffcf0:  00000005 00000005 00000008 3fff6230  
[12:41:54]3ffffd00:  00000005 00000000 00000020 4010098c  
[12:41:54]3ffffd10:  00000000 40243893 00000005 40101ed4  
[12:41:54]3ffffd20:  3ffebcf5 4010517f 3ffef708 3ffef3e8  
[12:41:54]3ffffd30:  40102a6b 3ffef708 3fff6204 3fff6204  
[12:41:54]3ffffd40:  ffffffe6 00e5d632 3fff018c 40102c48  
[12:41:54]3ffffd50:  3ffec5a0 00000000 00000000 4010098c  
[12:41:54]3ffffd60:  ffffffe6 00e5d632 4010310a 00000100  
[12:41:54]3ffffd70:  3ffec5a0 7fffffff 00000000 00000001  
[12:41:54]3ffffd80:  00000001 00000080 00000020 4010098c  
[12:41:54]3ffffd90:  00000005 00000000 00000020 4010098c  
[12:41:54]3ffffda0:  40102305 40104074 00000005 40101ed4  
[12:41:54]3ffffdb0:  3ffebcf5 4010517f 3ffef6e0 00040000  
[12:41:54]3ffffdc0:  40102a6b 3ffef6e0 3ffefeb8 3ffefeb8  
[12:41:54]3ffffdd0:  00000005 00000000 00000020 4010098c  
[12:41:54]3ffffde0:  3ffec5a0 00000000 00000005 40101ed4  
[12:41:54]3ffffdf0:  3ffebcf5 4010517f 3ffef6e0 00000100  
[12:41:54]3ffffe00:  40102a6b 3ffef6e0 00002200 00000001  
[12:41:54]3ffffe10:  00007fff 00e648b8 3fff018c 40102c48  
[12:41:54]3ffffe20:  3ffec5a0 00000000 00000000 00e635c8  
[12:41:54]3ffffe30:  00007fff 00e648b8 4010310a 00000100  
[12:41:54]3ffffe40:  3ffec5a0 7fffffff 00002200 00000001  
[12:41:54]3ffffe50:  00000001 00004a08 00000010 ffffffff  
[12:41:54]3ffffe60:  3ffec5a0 3fff32d8 00000000 00e648b8  
[12:41:54]3ffffe70:  3ffec3d8 2c9f0300 4000050c 3fffc278  
[12:41:54]3ffffe80:  40102e2c 3fffc200 00000022 3fff32a4  
[12:41:54]3ffffe90:  40231de0 00000030 00000010 ffffffff  
[12:41:54]3ffffea0:  40231e29 00000000 00000000 00000000  
[12:41:54]3ffffeb0:  3fff1c74 00000000 00000000 fffffffe  
[12:41:54]3ffffec0:  00000000 3fffc6fc 00000000 3fff2758  
[12:41:54]3ffffed0:  00000000 3fffdad0 3fff2798 00000030  
[12:41:54]3ffffee0:  00000000 00000000 40100b84 00003af3  
[12:41:54]3ffffef0:  00000000 559555cd 40100b84 00003af3  
[12:41:54]3fffff00:  00000000 4bc6a7f0 f0a3d70a 00000000  
[12:41:54]3fffff10:  00000000 00000000 4bc6a7f0 00000000  
[12:41:54]3fffff20:  00000001 3fff24c0 40100b84 00003af3  
[12:41:54]3fffff30:  00000000 00000000 3fff32a4 40223cf4  
[12:41:54]3fffff40:  00442a0e 3fff2078 40100b84 00003af3  
[12:41:54]3fffff50:  007a1200 5597cfa8 00000000 3fff2798  
[12:41:54]3fffff60:  00000000 00000000 00000001 4010098c  
[12:41:54]3fffff70:  3fff573c 40221358 3fff2758 00000000  
[12:41:54]3fffff80:  3fffdad0 3fff2798 40231e3b 3fffefa0  
[12:41:54]3fffff90:  3fffdad0 00000000 3fff2758 4021f6e0  
[12:41:54]3fffffa0:  feefeffe feefeffe feefeffe 40231f24  
[12:41:54]<<<stack<<<
[12:41:54]
[12:41:54]last failed alloc call: 4021A939(152)
[12:41:54]
[12:41:54] ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[12:41:54]
[12:41:54]load 0x4010f000, len 1384, room 16 
[12:41:54]tail 8
[12:41:54]chksum 0x2d
[12:41:54]csum 0x2d
[12:41:54]vbc204a9b
[12:41:54]~ld
[12:41:54]Ada
[12:42:05]
[12:42:05]Exception (29):
[12:42:05]epc1=0x402278d5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000000ad depc=0x00000000
[12:42:05]
[12:42:05]>>>stack>>>
[12:42:05]
[12:42:05]ctx: sys
[12:42:05]sp: 3fffecc0 end: 3fffffb0 offset: 01a0
[12:42:05]3fffee60:  3fff2fb8 3fff1f98 3fffbe84 40227f67  
[12:42:05]3fffee70:  3fffbe84 3fff2f24 3fff2f20 40227fac  
[12:42:05]3fffee80:  3fff2f34 3fffbb24 3fff2fac 4025c452  
[... 262 lines cut ...]
[12:42:06]3ffffef0:  00000000 000045ff 40100b84 00002a5a  
[12:42:06]3fffff00:  00000000 4bc6a7f0 3b645a1c 00000000  
[12:42:06]3fffff10:  00000000 00000000 4bc6a7f0 00000000  
[12:42:06]3fffff20:  00000001 3fff24c0 40100b84 00002a5a  
[12:42:06]3fffff30:  4025f84e 00000000 3fff2297 40231e63  
[12:42:06]3fffff40:  40260a2b 00000000 3fff35ac 40222563  
[12:42:06]3fffff50:  3fff21f0 3fff1f60 00000000 4021ef4e  
[12:42:06]3fffff60:  00000000 00000000 00000001 4010098c  
[12:42:06]3fffff70:  00000000 3fffff60 3fff2758 00000000  
[12:42:06]3fffff80:  3fffdad0 3fff2798 40231e3b 3fffefa0  
[12:42:06]3fffff90:  3fffdad0 00000000 3fff2758 4021f66b  
[12:42:06]3fffffa0:  feefeffe feefeffe feefeffe 40231f24  
[12:42:06]<<<stack<<<
[12:42:06]
[12:42:06]last failed alloc call: 40227F67(216)
[12:42:06]
[12:42:06] ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[12:42:06]
[12:42:06]load 0x4010f000, len 1384, room 16 
[12:42:06]tail 8
[12:42:06]chksum 0x2d
[12:42:06]csum 0x2d
[12:42:06]vbc204a9b
[12:42:06]~ld
[12:42:06]Ada

Woutch, 2020-04-18:
Same issue as I have: #848

See: https://github.com/home-assistant/core/issues/33921

triggerx, 2020-04-22:
I'm having this issue with 0.9.1, and I don't even use HA.  No problems with 0.8.6... nothing BUT problems with 0.9.x.

closed 2020-08-28 by stale[bot]