wled/WLED #859: Cannot upload code through USB port

sjude68, 2020-04-21:
I know this is not the place to ask but it would be nice if anyone can help.

I have been uploading wled code using USB port to D1 Mini after compiling using Arduino.  But recently I made a mistake connecting the 5V & GND from a DC power source after uploading the code.  Because of this I cannot upload using USB port.  The D1 Mini works fine with wled and the 4-relays with the modified code.  I was able to re-upload using Manual OTA.  But now I cannot upload because OTA is switched off.

Can anyone help me to upload a new code using the TX & RX pins from another D1 Mini or USB Serial adaptor.  The D1 Mini is working fine when connected to 5V power source but code is not uploading through the Serial USB adaptor.  I have used the adaptor to upload code to Arduino Mini which connecting TX, RX, Vcc. Gnd & DTR pins to the Serial adaptor.  But the D1 Mini does not have a DTR pin.  How do I put the D1 Mini to loading mode.

Any help would be appreciated so that I can use it again for Wled.

Aircoookie, 2020-04-21:
Hi,
you probably blew the USB current-limiting diode by connecting the 5V power source and USB simultaneously. It should now be safe to power the ESP via DC while uploading new code via USB. Alternatively, you can desolder the diode shown below and bridge the contacts, but be very careful if you do that as you might damage your computer if you connect usb and dc again at the same time.

If you want to attempt flashing via an external serial adaptor, tie gpio0 (D3) to ground to enter programming mode. You might also need to press the reset button at some point during the flashing process. Hope it helps!

Aircoookie, 2020-04-21:
Sorry forgot the image with the diode: 
[image]

austwhite, 2020-04-21:
The other option, if you really did get the USB, is to try a USB programming adaptor like is used for esp-01 and just use jumper wires to the correct pins

sjude68, 2020-04-21:
I have another Nodemcu which I fried near the Vin port. It was by reverse polarity while applying 5v power to it. I am not sure how I can retrieve it. Usb port doesn't work. Besides Vin pin where can I power it up. Hope it's not completely dead. I kept it with me hoping to get it working again.

I can see the burnt up rectifier near Vin.

austwhite, 2020-04-21:
If you reverse polarity it on the 5V you could try a 3.3V power source between gnd and 3.3V.
This is a long shot though.  They are not very tolerant of reverse polarity.

sjude68, 2020-04-21:
I will try that option and post my results.

sjude68, 2021-04-04:
Sorry to ask again. Hope to get help here itself.

Can anyone give me the part number for the diode and the 5v regulator. I am not finding the correct part as there are various available which takes variable input voltages from 1.8v - 5.5v. Can I use this part.

Aircoookie, 2021-04-06:
Wemos D1 mini seems to use B5819w diode and ME6211 5v-3.3v regulator according to the datasheet.

I don't know if it would be economical to fix the board though, considering shipping for the components.

sjude68, 2021-04-07:
Thanks Aircoookie/WLED for the reply. I will keep this information. I will
get a new D1 mini instead.

closed 2021-04-07 by sjude68