wled/WLED #864: Feature request: Presets to dedicated pins on button.

craigrobbo, 2020-04-22:
Is it possible to activate vertain presets on buttons (more than one)

Also the possibility of the option 1 that a momentary press would activate preset x while holding and revert back to previous when released.

a second option (switchable in the interface) so that the button can just change the preset when pressed.

JoshWobbles, 2020-04-30:
Yes please! especially the cycle option. I have permanently mounted mine and would like to have a button that allows me to cycle my presets without having to pull out my phone!

Aircoookie, 2020-05-04:
Sure! Just go to time and macro settings, and save `P1=1&P2=4&PL=~` in a macro slot. This will cycle through the presets 1-4. Now just set the button short press macro to the number of the macro you used. You can also assign macros to double and long click, for example `T=2` for on/off or `PL=1` for loading the first preset :)

Aircoookie, 2020-09-02:
@JoshWobbles hi again! Did my post from May 5th not cover your use case (cycling presets with a button press)? If so, I must have misunderstood, please explain again what exactly you would like the button to do :)

maximbeliveau, 2020-12-24:
I Have the same question ! i think he was trying to say, : Switch the preset on button press, so each time you press the button the preset change until you come back to the top of the list and start again.

bstein2379, 2020-12-30:
@maximbeliveau it works just as @Aircoookie described above on May 5th. Here is what I did. I have 15 saved presets, so macro 1 will cycle through each preset 1 through 15 and then back to 1 for each single button press. A double press of the button will toggle the LEDs on/off.

Go to Config --> Time & Macros. Under Advanced Macros put...
1: **P1=1&P2=15&PL=~**
2: **T=2**

Then put...
Button short press macro: Macro: **1**
Double press: **2**

Don't forget to hit **Save**.

If you don't have 15 presets, just put how many you have for P2=**x** on macro 1.

You can use this page to understand all the different supported macro configurations in case you want to customize it in any way you want.
https://github.com/Aircoookie/WLED/wiki/HTTP-request-API

closed 2021-06-16 by stale[bot]