wled/WLED #866: Wemos D1 Mini connect to wifi, only with usb power.

evertonluiz, 2020-04-22:
I tried to install wled several times on several different wemos, but I'm having a problem.
My setup has 34 leds, so I feed it with an external power source, which also feeds the wemos through the 5V and GND ports, but when powered only by the external power source it doesn't connect to the wifi, only the button works.
When i plug the usb cable wemos starts and connect to wifi.

Aircoookie, 2020-04-23:
Hi!
Do you have several working setups already, or did you try getting this one to work with multiple Wemos boards? I and many others have several working installations where the ESP is powered from the external source. I have had the exact same issue, but in reverse, the board wouldn't connect if I plugged in an USB cable without connecting it to a PC on the other side. You could also try a different power source (for 34 LEDs, I'd recommend a phone charger with an old USB cable stripped), maybe the one you are using is creating RF noise (although that shouldn't often be an issue...)

evertonluiz, 2020-04-23:
O tried a 2A and 4A Power source.

Aircoookie, 2020-04-23:
Can you send a pic of your setup? If you tried multiple power sources as well, I have no clue why it isn't working...

evertonluiz, 2020-04-25:
I found It! It was a problem with my wemos Lot, I bought another one, and it's ok

closed 2020-04-25 by evertonluiz