wled/WLED #867: Cannot switch from segment preset back to others

jaydawgx7, 2020-04-23:
I'm not sure if this is user error or just as far as the current development has gotten regarding segments, but I'm having issues leaving a segment preset (16).  I have other presets with normal patterns and colors set up, however, after using preset 16 with segments, I can't seem to get back to my others without rebooting.  When I tap on them, nothing happens. I can somewhat get around the issue by deleting the segments since they come back when i go to preset 16, however, this approach causes some weirdness like flickering.

Aircoookie, 2020-04-23:
Hi, yes I have to admit the current preset system is not ideal, that's why it is planned to be completely revamped shortly :)

If the other presets don't get applied at all, make sure a segment is selected (has a checkmark next to it in Segments tab) in the config stored in preset 16. If it doesn't, check one and then re-save preset 16. If you then load a different preset, its config should be applied to all segments that are selected.

jaydawgx7, 2020-04-24:
Thanks for the response, I'll give that a try! I'm really looking forward to your future work around presets and segments. I like assigning static colors to specific LEDs for a lot of my projects and it's great being able to set them up with my phone rather than writing some ardruino code every time I want to change some colors

closed 2020-08-29 by stale[bot]