wled/WLED #869: Pls add option to disable LED_BUILTIN (nodemcu)

ilyaroot, 2020-04-24:
Pls add option to disable LED_BUILTIN (led wi-fi connection?)
NodeMcu has rather bright LED, so it disturbs at night.

huggy-d1, 2020-04-24:
A quick solution is black electrical tape wrapped around the NodeMCU board.  Wrap it around the board just to the left of the LED.  Continue the wrap all the way around the board, and this time, just to the right of the LED.  Continue the wrap one more time, directly centered over the LED, and end it about 1/2 way around the back of the NodeMCU.  Do flip back a few mm (1/4") of tape onto itself so you can easily remove it later on.
You can recompile the WLED project and disable the builtin LED by defining the built-in LED pin as 0.

austwhite, 2020-04-24:
Or you could put the node mcu in a little case and make the whole install look more professional than just having a board and wires hanging around 🙂

ilyaroot, 2020-04-24:
Oh, that's such an engineer's way to handle all problems - by fixing them with electrical tape!
Ok, I'll try to recompile project and disable the builtin LED in code. Thanks anyway!

mjferro, 2020-04-24:
This is documented in the wiki:

How to turn off the bright blue onboard LED?

This LED can be very distracting. Unfortunately it can't easily be disabled as it shares the gpio2/D4 pin with the LED output. Currently there are 3 workarounds:

Cover the LED

Remove the LED permanently (desolder or apply pressure with e.g. a flathead screwdriver)

Use a different LEDPIN, although the default is recommended for stable operation

JoshWobbles, 2020-04-30:
Personally I got a waterproof enclosure like this for mine as mine are permanently mounted anyway and I didn't want a loose board hanging around. https://amzn.to/3aY47W0

closed 2020-09-05 by stale[bot]