wled/WLED #871: Using E1.31 or UDP to send text to WLED

DrMarlin, 2020-04-25:
I have multiple WLED setups and love it. Most recently I've made a few RGB Matrixes - I can use Jinx to drive them via E.131 and it works well.

I'd ideally like a non-Windows solution that I can run headless (e.g. on a Pi or Linux VM) to control what is displayed on the matrix. LEDFX is great for visualizers, but is anyone aware of an option that could send text - e.g. a clock, messages, scrolling text etc. to WLED via E1.31 or UDP sync?

Aircoookie, 2020-04-25:
Hi!

Awesome that you like WLED!
xLights is a great open source application that has this capability and Linux versions available, although I find it more diffult to operate compared to Jinx.
Although I am not sure whether it is capable of headless operation, you can give it a try. It also supports output to multiple devices at once and can handle many advanced effects beyond text display!

pille, 2020-04-26:
> I'd ideally like a non-Windows solution that I can run headless (e.g. on a Pi or Linux VM) to control what is displayed on the matrix. 
>[...] is anyone aware of an option that could send text - e.g. a clock, messages, scrolling text etc. to WLED via E1.31 or UDP sync?

QLC+ does work headless and on a Pi and can render text on a matrix.
It just seems a little too complex for this (simple) task.

KekieM, 2020-04-28:
With xlights you can create files than can be played headless on a Pi with falcon player.

closed 2020-09-03 by stale[bot]